letscontrolit/ESPEasy #2199: P024_MLX90614 uses ports=16

Grovkillen, 2018-12-27:
https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P024_MLX90614.ino

We shouldn't use the port setting for this hardware, it's set to 0x5A (90) and is not changeable.

```
047: Device[deviceCount].Ports = 16;
...
112: byte unit = Settings.TaskDevicePort[event->TaskIndex];
113: uint8_t address = 0x5A + unit;
```

should be?
```
047: Device[deviceCount].Ports = 0;
...
113: uint8_t address = 0x5A;
```

I don't have this unit close by but will try to test my change this week.

ironataerail, 2018-12-27:
Datasheet shows no sign of multiple units in the same bus

https://www.sparkfun.com/datasheets/Sensors/Temperature/MLX90614_rev001.pdf

Also the device has only the 4 basic pins so I do not see how and address can be set.

BUT 

A quick google search resulted into a : "how to assign different slave address other than the default 0x5A"

https://chrisramsay.co.uk/posts/2014/09/arduino-and-multiple-mlx90614-sensors/

So I guess one could have up to 127 units on the same bus if wanted to push the envelop.

-D

Domosapiens, 2018-12-27:
Port is used as an offset to 0x5A (90).
So in general ... leave it 0 (no offset) .. is my experience.

ironataerail, 2018-12-27:
My vote is to remove it (anything else is just asking for trouble)

ShardanX, 2018-12-31:
Address seems to be stored in an EEPROM, writeable.
http://forum.arduino.cc/index.php?topic=54170.0
https://chrisramsay.co.uk/posts/2017/09/arduino-and-multiple-mlx90614-sensors-take-two/

Domosapiens, 2019-01-02:
Address change ...something as seems possible (??) with _P047_i2c-soil-moisture-sensor.ino
`// check if we want to change the sensor address
        if (Settings.TaskDevicePluginConfig[event->TaskIndex][4]) {
        	addLog(LOG_LEVEL_INFO, String(F("SoilMoisture: Change Address: 0x")) + String(_i2caddrP47,HEX) + String(F("->0x")) +
        			String(Settings.TaskDevicePluginConfig[event->TaskIndex][3],HEX));
        	if (Plugin_047_setAddress(Settings.TaskDevicePluginConfig[event->TaskIndex][3])) {
        	  Settings.TaskDevicePluginConfig[event->TaskIndex][0] = Settings.TaskDevicePluginConfig[event->TaskIndex][3];
        	}
        	Settings.TaskDevicePluginConfig[event->TaskIndex][4] = false;
}`