letscontrolit/ESPEasy #2200: If plugin don't have pullupoption + inverselogicoption the sensor settings end up on task settings

Grovkillen, 2018-12-27:
If a plugin have these settings set to false the next sensor setting will end up on Task settings.
```
        Device[deviceCount].PullUpOption = false;
        Device[deviceCount].InverseLogicOption = false;
```
Will output this:
[image]

```
        Device[deviceCount].PullUpOption = true;
        Device[deviceCount].InverseLogicOption = false;
```
Will output this:
[image]

TD-er, 2018-12-27:
Can you explain a bit more what's going on?
I don't see it

Grovkillen, 2018-12-27:
Look at the altitude input. It's part of the task settings section. If I enable the pull-up (or over är inverted logic) it jumps down to the sensor section (where it belongs).

Grovkillen, 2018-12-28:
We should rename it to Device instead of Sensor.

Grovkillen, 2019-10-10:
Is this something we need to fix?

TD-er, 2019-10-10:
> Is this something we need to fix?

Let's ask the one that created the issue ;)
Is it something that's still an issue then yes.

tonhuisman, 2022-05-07:
This seems outdated

closed 2022-05-07 by TD-er