letscontrolit/ESPEasy #2201: a base directory for esp32

haraldtux, 2018-12-28:
It's an idea to

a base directory for esp32 in to the source directory with

* boot_app0.bin 
* bootloader.bin
* ESPEasy.ino.partitions.bin

with a small description and example (esptool) script ?

I think you there are many people there with a fun does

Grovkillen, 2018-12-28:
Please explain more, it's a bit fragmented.

haraldtux, 2018-12-28:
I mean a directory with files and a small description 
for first step to get run a ep32

If you are like me it is very difficult to have difficulty reading that on a forum and or internet to find out all

Sorry, I'm not that good to describe things and in English

Grovkillen, 2018-12-28:
Ok, but we are not officially pushing ESP32 firmware at this time. If you want to test it out you are left a bit on your own. We must focus on the release of 2.1 for ESP82XX micro controllers. Later versions will have official support for the ESP32 but we are not there yet.

Grovkillen, 2018-12-28:
I will close this since its not really suitable for the issue tracker... please use the forum (even if you don't like it).

closed 2018-12-28 by Grovkillen

haraldtux, 2018-12-28:
ok, thank you, is good