letscontrolit/ESPEasy #2202: Nodemcu-32s (aka. ESP-WROOM-32) crash:  flash read err, 1000 ets_main.c 371

manhere, 2018-12-28:
Flash lastest release "ESP_Easy_mega-20181220_esp32dev.bin"
Board:  Nodemcu-32s (aka. ESP-WROOM-32)
crash log loop as below:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

Grovkillen, 2018-12-28:
If you intend to use the ESP32 we urge you to be able to supply more testing and not only bug reports. We're not officially supporting the ESP32 since all our time goes to the release of 2.1 which is only supporting ESP82XX units... later versions will have support for ESP32 but we don't have time to investigate vague reports like this. I hope you understand.

If you want to be a tester then you need to add much more info and more testing. A proper report in other words 👍

redskinhu, 2018-12-29:
Hi

Is your GPIO-12 high or low on boot? I remember this flash read error (maybe your reason is different). If the GPIO-12 is high on boot the flash get only 1.8V and the module dont't boot.
See: [https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#other-pins](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#other-pins)

 If you need GPIO-12 high on boot you can burn the relevant efuses and the flash on the module get 3.3V always.
See: [https://github.com/espressif/esptool/wiki/espefuse#setting-flash-voltage-vdd_sdio](https://github.com/espressif/esptool/wiki/espefuse#setting-flash-voltage-vdd_sdio)

I hope it helps.

T.

TD-er, 2018-12-30:
See also this page in the documentation: [Best pins to use on ESP32](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp32)

voducngh, 2018-12-31:
> Flash lastest release "ESP_Easy_mega-20181220_esp32dev.bin"
> Board: Nodemcu-32s (aka. ESP-WROOM-32)
> crash log loop as below:
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371
> ets Jun 8 2016 00:22:57

Hello @manhere 

You need to flash the bootloader. So under Windows10/cmd:

1. I do erase:  flasherase.cmd:
esptool.exe --chip esp32 --port COM11 --baud 256000 --before default_reset --after hard_reset erase_flash

2. I do use flash the boot loader: flashbootloader.cmd:
esptool.exe --chip esp32 --port COM11 --baud 256000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x8000 partitions2.bin 0xe000 boot_app0.bin 0x1000 bootloader.bin

3. and flashfirmware.cmd:
esptool.exe --chip esp32 --port COM11 --baud 256000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x10000 firmware.bin

Then I connect my PC to the ESP-32's AP, and configure the device ESP32 to connect to my Wifi network.

FullMetalRico, 2019-05-09:
> You need to flash the bootloader. So under Windows10/cmd:

>1.  I do erase: flasherase.cmd:
> 2. I do use flash the boot loader: flashbootloader.cmd:

This is the only thing of all the "solutions" i tried that actually made things better. Thank you very much for it. 
After the first 2 commands i could upload my sketches to the ESP32 via the Arduino IDE without any Problems and the error did not occur again for 3 days now. Before, that bs came every 15 minutes or so.

PS: If you have never flashed the ESP32 on windows without the Arduino IDE before make sure you install Python 2.7.X instead of 3.X before getting the esptool.py via pip and doing the steps.

adri94, 2020-01-23:
Hello,

Same issue here.
Where to find : partitions2.bin ; boot_app0.bin ; bootloader.bin for my esp32 ?
It is for espeasy.

Thx,

FullMetalRico, 2020-01-23:
> Where to find : partitions2.bin ; boot_app0.bin ; bootloader.bin for my esp32 ?

You can find those files in your esp32 firmware. Just download an AT Firmware for your esp32 from the official espressif website. When i made that i downloaded the version "ESP32-WROOM-32 AT Bin V1.1.2" but i gues you can choose any version as long as it is for the esp32-wroom. I do not know if i am allowed to post a link but here it is. Just download it, unpack and search for the needed files.

https://www.espressif.com/en/support/download/at?keys=&field_type_tid%5B%5D=13

adri94, 2020-01-23:
Many thanks ! 
I found it on the espEasy32 page.
I can make the ESPEasy32_R20100 version work but any other version from ESPEasy_mega (more recent) does not work. 
Any idea ?

FullMetalRico, 2020-01-24:
> Any idea ?

Sadly not, because i am not working with espEasy. I'm more of a do-it-yourself programmer :)

tonhuisman, 2023-09-02:
Support for ESP32 and related CPUs has been greatly enhanced, so this issue can be closed.

closed 2023-09-02 by TD-er