letscontrolit/ESPEasy #2206: Add new Nodetype RPIEasy

enesbcs, 2018-12-30:
### Summarize of the problem/feature request

Please add RPIEasy as a Node Type into the ESPEasy P2P network.

I think the changes below will be enough.

Target source file1: "ESPEasy-Globals.h"
#define NODE_TYPE_ID_RPI_EASYM_STD         5

Target source file2: "WebServer.ino"

void handle_root()
...
            case NODE_TYPE_ID_RPI_EASY_STD:
              TXBuffer += F("RPI Easy");
              break;
...

void handle_json()
...
            case NODE_TYPE_ID_RPI_EASY_STD:    platform = F("RPI Easy");     break;
...

enesbcs, 2018-12-30:
I've made some CTRL-C mistake:
#define NODE_TYPE_ID_RPI_EASY_STD 5

TD-er, 2018-12-30:
Do you also have a preferred URL to the project, which can be set as comment next to the definition?

enesbcs, 2018-12-31:
Maybe the github URL ( https://github.com/enesbcs/rpieasy ), but not strictly necessary. Thank you!

TD-er, 2019-01-06:
Added it and also moved some string lookup functions to a central place + removed some code duplication.

closed 2019-01-06 by TD-er