letscontrolit/ESPEasy #2208: Can not Build any custom ESPEasy that works on ESP8266-01 modules

petefirefox, 2019-01-02:
Hello everyone,
I have found that in the define_plugin_sets.h the plugin for ID12 (P040) is not included, allthough it is announced as a default plugin.
I need this plugin for my project of a RDM6300 RFID reader based door system.

So I try to do a custom build based on the latest release of 20181231.

Unfortunately I get the famous boot loop with my platformio build attempts and in Arduino the devices do not respond at all after flashing. The ESP8266-01 module I use has the famous PUYA chip on it.

Would it be possible to include the P040 plugin in the regular builds? That would make my life a lot easier.
Thanks a lot.

Peter

TD-er, 2019-01-02:
If you're using core 2.5.0, then the puya patch is already merged in the core libraries.
For older core libraries we patch the libraries before building them, which is rather hard to do under Windows.
So simply try to use the core 2.5.0 via platformio.ini.

petefirefox, 2019-01-02:
Hello Gijs,

I can actually confirm that the official builds work fine for me. But they do not contain the support for RDM6300 plugin P040. If that would be fixed/addressed, then I could actually switch to those official builds and would be very (extremely) happy.

Would it be possible to correct the define_plugin_sets.h? Thanks.

Peter

petefirefox, 2019-01-02:
Hello Gijs,

you made my day! Thanks a lot. The hint for core 2.5 was the perfect hit.
Now I made a custom build with the P040 plugin included and works perfect fine.

Case can be closed.

Peter

TD-er, 2019-01-02:
Glad to hear [this patch](https://github.com/esp8266/Arduino/pull/5504#issue-238919838) is working fine :)

closed 2019-01-02 by TD-er