letscontrolit/ESPEasy #2209: ESP.Easy.Flasher.exe is been seen as Virus by three Virus Engines?

DENightOne, 2019-01-03:
latest release ESPEasy_mega-20181231 the file ESP.Easy.Flasher has a virus....in all previous versions it was scanned as virus free.  VirusTotal also reports the file as a virus but only the version included in the ESPEasy_mega-20181231 release

DENightOne, 2019-01-03:
Thanks....I thought it might be a false positive.

uzi18, 2019-01-03:
What kind of virus is found there?

Grovkillen, 2019-01-03:
I will add info about it in the read me file though. Thanks for the feedback

Grovkillen, 2019-01-03:
Thank you for that! 😎👍

What did the virus SW report about the file? I'm gonna update it soon to let the users be able to download latest releases without the need to go to GitHub.

DENightOne, 2019-01-03:
https://www.virustotal.com/#/file/d66dd9f59f131c5575b9ee2db84eb09cd5665a12bd5cd3b496475eda7919ad7c/detection

Should give you idea of the behaviour...also Microsoft Defender picked it up as

https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?name=Trojan%3aWin32%2fCloxer.D!cl&threatid=2147726003

Hope this helps

ShardanX, 2019-01-03:
FYI
Just checked with Avira - no result

TD-er, 2019-01-03:
When in doubt, you can always run the file through this: [Jotti Virusscanner](https://virusscan.jotti.org/nl)

pmboothby, 2019-04-01:
FYI, Bitdefender just quarantined ESP.Easy.Flasher mega-20190315 when it had nearly finished flashing an ESP-01. BTW, thanks for the great tool!