letscontrolit/ESPEasy #2211: MQTTimport not work?

itProfi, 2019-01-05:
[image]
[image]
[image]
[image]
[image]
[image]

itProfi, 2019-01-05:
What i do wrong? Absolutly nothing happing...

agsochi, 2019-01-05:
Hi, you subscribed in MQTTImport on `cmdMQTT` topic, but right topic is `sonoff_basic_bath2/cmdMQTT`

Grovkillen, 2019-01-05:
Yes just like @agsochi says. The setup you're using is to be able to send to ALL units with one message. Try publish on topic `cmdMQTT` and you'll see.

itProfi, 2019-01-05:
> sonoff_basic_bath2/cmdMQTT

Yea! You are absolutly right! But why there is Subscribe topic in controller settings? Why in wiki no about this?? :)
[image]

> Yes just like @agsochi says. The setup you're using is to be able to send to ALL units with one message. Try publish on topic `cmdMQTT` and you'll see.

No. Work only as say @agsochi

itProfi, 2019-01-05:
Please, update Wiki!!

Grovkillen, 2019-01-05:
To clarify, if publishing to `cmdMQTT` all units with MQTT import subscribing to this topic will fetch this message.