letscontrolit/ESPEasy #2214: Uptime is logged twice via syslog

fly74, 2019-01-06:
maybe since 20181231, also with 0106

```
06.01.2019 07:39 | EspEasy: Webserver: start
-- | --
06.01.2019 07:39 | EspEasy: NTP : NTP replied: delay 21 mSec Accuracy   increased by 0.065 seconds
06.01.2019 07:39 | EspEasy: Time adjusted by -1546756778332.14 msec. Wander:   -429654660.65 msec/second
06.01.2019 07:39 | EspEasy: Current Time Zone: STD time start: 2019-10-27   03:00:00 offset: 60 min
06.01.2019 07:39 | EspEasy: EVENT: Time#Initialized
06.01.2019 07:39 | EspEasy: EVENT: Clock#Time=Sun,07:39
06.01.2019 07:40 | EspEasy: EVENT: Clock#Time=Sun,07:40
06.01.2019 07:40 | EspEasy: WD : Uptime 1 ConnectFailures 0 FreeMem 27024
06.01.2019 07:40 | EspEasy: WD : Uptime 1 ConnectFailures 0 FreeMem 27048
06.01.2019 07:41 | EspEasy: EVENT: Clock#Time=Sun,07:41
06.01.2019 07:41 | EspEasy: WD : Uptime 2 ConnectFailures 0 FreeMem 27048
06.01.2019 07:41 | EspEasy: WD : Uptime 2 ConnectFailures 0 FreeMem 27048
06.01.2019 07:42 | EspEasy: EVENT: Clock#Time=Sun,07:42
06.01.2019 07:42 | EspEasy: WD : Uptime 3 ConnectFailures 0 FreeMem 27048
06.01.2019 07:42 | EspEasy: WD : Uptime 3 ConnectFailures 0 FreeMem 27048
06.01.2019 07:43 | EspEasy: EVENT: Clock#Time=Sun,07:43
06.01.2019 07:43 | EspEasy: WD : Uptime 4 ConnectFailures 0 FreeMem 27048
06.01.2019 07:43 | EspEasy: WD : Uptime 4 ConnectFailures 0 FreeMem 27048
06.01.2019 07:44 | EspEasy: EVENT: Clock#Time=Sun,07:44
06.01.2019 07:44 | EspEasy: WD : Uptime 5 ConnectFailures 0 FreeMem 27048
06.01.2019 07:44 | EspEasy: WD : Uptime 5 ConnectFailures 0 FreeMem 27048
06.01.2019 07:45 | EspEasy: EVENT: Clock#Time=Sun,07:45
06.01.2019 07:45 | EspEasy: WD : Uptime 6 ConnectFailures 0 FreeMem 27048
06.01.2019 07:45 | EspEasy: WD : Uptime 6 ConnectFailures 0 FreeMem 27048
06.01.2019   07:46 | EspEasy: EVENT: Clock#Time=Sun,07:46

```

Grovkillen, 2019-01-06:
It's been this way since always. It's intended. Does it cause a problem?

We use it to indicate the 500msec point in the loop.

TD-er, 2019-01-06:
> We use it to indicate the 500msec point in the loop.

Nope, it is printed in the 30 sec loop. :)
We don't have a 60 second loop.

Grovkillen, 2019-01-06:
Of course!! 😋 Too early...

TD-er, 2019-01-06:
We have several loops, each handling different tasks.
In short, there is a loop for:
- 50-times/sec
- 10-times/sec
- 1-times/sec
- one time per 30 seconds.

And some other ones too.
Each plugin has also the option to have function calls implemented for things to do in these loops.

fly74, 2019-01-06:
Why the 30 seconds loop do a syslog entry?

Grovkillen, 2019-01-06:
Because if not we need to add a 60 second loop. Is it causing problems?

closed 2019-01-06 by Grovkillen

TD-er, 2019-01-06:
> Why the 30 seconds loop do a syslog entry?

It is some kind of continuous statistics, logged at LOG_LEVEL_INFO