letscontrolit/ESPEasy #2220: oledframed field content length too short

netraam3371, 2019-01-07:
Hi all,
It seems the oledframed lines are limited to a very small amount of characters. For normal text, this isn't so much of a problem, but using the default entity names as "Temperature" for a temp/humidity sensor, there is not enough room to have a simple temp value on the display using a [sensorname#valuename] entry.
Hope this can be easily updated.
Side note: it'd be nice to have the scrolling between display pages go in the right direction. Next pages slide in from the left instead of from the right, which makes it reverse for left-to-right text.
And being able to control the header fields would also be nice. Node name and WLAN network name could be a sane default, but I can think of quite some other sensible values. Node name could make sense for developers or power users, but often not for end users just viewing the display.

TD-er, 2019-01-07:
It is better to have those names short, also when using them in rules.
Changing the size of stored values isn't very easy to do and still keep compatibility with existing installs.
Not sure how easy the scroll direction can be changed.
I have to dig into the library to see if that's doable.
The assignment of texts in upper and lower lines has been requested before

Grovkillen, 2019-01-07:
Just as a side node. I have never seen a temperature display stating "temperature" in front of the temperature value ;)

TD-er, 2019-01-07:
@Grovkillen but most plugin variable names have long names while `T` would also be clear.

Grovkillen, 2019-01-07:
Yeah, and we should try to uniform the way users name the tasks etc... it would be easier to copy settings from examples etc.

TD-er, 2019-01-07:
We can change the default names for all plugins

Grovkillen, 2019-01-07:
We should. But then we also need to have a number (the task number?) to not make the default collide with some other task.

TD-er, 2019-01-07:
The value names can be shortened.

netraam3371, 2019-01-07:
As mentioned by TD-er, it's not about displaying the word "Temperature" but using the [TaskName#ValueName] pairs in the fields used to display the data, that tend to take much more space in the definition than on the screen.

TaskName shouldn't be overly long, but eg. mh-z19b-3 (for identifying a specific sensor number) isn't that long in my opinion. Shortening that to z19b3 just to save a few bytes is possible but would make it harder to process and compare it to values from mh-z19b-1 and -2 that happen to not have a display connected to the node.

Default ValueNames could be shortened from Temperature to Temp or even T, but when selecting a larger number of lines to be displayed (and thus resulting in a smaller font size) we can probably display like 15-20 chars per line. For a displayed line to be useful, I assume it would have at least one [task#value] reference, so isn't the current max length of the fields on the short side anyway?

Grovkillen, 2019-01-07:
Ah, sorry for not understanding that.

TD-er, 2019-01-07:
The way things are stored in our setup is not making it very easy to change the size of a stored field.
So it may indeed be short right now, but it will be relatively hard to change afterwards.

And the sensor could also be named "co2", since that's what you're measuring and you will probably only have one on a board.

netraam3371, 2019-01-07:
I have multiple mh-z* sensors that have some kind of offsets and anomalies (luckily the senseair s8 behaves much better), and to rule out any power issues or connection issues, I rotate these over a set of nodes. So I need somewhat descriptive names here. Could be co2a, co2b, co2c, etc. but I was hoping something better was possible as it's no problem for just uploading the measurements but only when trying to display the info on nodes with display module.

But if changing the field length is a major operation, I agree it would be wiser to postpone that till other major maintenance is done on the oledframed plugin. (Such as making other settings configurable, which would probably change the setup format anyway.)

PR #3050 (2020-07-23): [P036] Memory optimization (issue #2799) and other improvements
CHG: Memory optimization (issue #2799)
CHG: Added setting for 'Disable scrolling while WiFi is not connected',
         scrolling is not smooth as long as the ESP tries to connect Wifi (Load 100%!)
BUG: Wifi-Bars not fully right for 128 pixel wide displays
BUG: clearing window for scrolling lines was too high and to narrow
BUG: content of lines with pixel width > 255 had to be shortened
CHG: scrolling lines improved
CHG: variables renamed for better understanding

Fixes: #2312
Closes: #3026 
Fixes: #3127 
Fixes: #2760 
Fixes: #279 [... cut]
closed 2020-07-23 by TD-er