letscontrolit/ESPEasy #2225: GPS config produces crash

Wiki591, 2019-01-08:
For curiosity I flashed a test device (Wemos D1) with ESP_Easy_mega-20190108_test_ESP8266_4096.bin to have a look at the new GPS plugin. When I went into the GPS configuration (no serial device connected, serial not disabled in advanced config) the system crashed at the moment I choose a topic in the pulldown for "Software/Hardware Serial". 

It was no more possible to enter the configuration page, the system crashes immediately at the moment the pulldown should appear on the browser.

Grovkillen, 2019-01-08:
This is known, thanks for reporting it though! 👍

Wiki591, 2019-01-08:
Yes, I was accidentally misbehaving from the logical side - but this may happen in the future to other users.

I encountered the problem, that after my misbehaving it was not possible to remove the device. Opening the config page resulted in a reboot of the system. Thats why I tried the factory reset :-)))

Probably there should be a "delete" button in the overview of the attached devices? I mean, at current state, if you delete a device out of its config page, you get the error message "SaveTaskSettings taskIndex does not match".

TD-er, 2019-01-08:
The options to delete should indeed be extended.
If I'm correct there has been at least one issue report about it, but still it hasn't been added.

TD-er, 2019-02-25:
This one should have been fixed.
Could you please verify and close the issue if it has indeed been fixed?

Wiki591, 2019-02-26:
Yes, I will do it on Saturday when I am back home

Wiki591, 2019-03-05:
Tested with no device connected. Now its possible to choose the different hw serial configurations without a crash, removing of the device works, too.
Thanks for taking care.

closed 2019-03-05 by Wiki591