letscontrolit/ESPEasy #2226: Factory reset doesn't keep WIFI credentials

Wiki591, 2019-01-08:
Tried out to do a factory reset keeping all possible five configs with ESP_Easy_mega-20190108_test_ESP8266_4096.bin.

After reboot the device started the AP mode, I entered using serial console the WIFI credentials and afterwards everything was fine - the other configs have been kept exept Wifi.

Grovkillen, 2019-01-08:
Maybe we should have it optional.

Wiki591, 2019-01-08:
Well, its already optional. I've activated it, the other settings were kept perfectly, but not the Wifi. For me personally its OK as it is. Keeping the Wifi credentials, too, would be the cherry on the cheesecake.

Btw.: genius functionality, indeed, thanks for that.

Domosapiens, 2019-01-08:
Functionality was introduced with #2083
@Grovkillen : I think it deserves the label "bug"

TD-er, 2019-01-08:
I've seen this happen before and if you look at the serial log when this happens, it will probably show you that it could not find any valid wifi credentials and thus start the AP mode.
Not sure why that happens, since I do save it after format SPIFFS and creating new settings.
It doesn't happen always. So maybe the last write doesn't flush the settings?
The credentials are stored in another file in the SPIFFS, so maybe we should unmount the filesystem first before reboot?

commit f77b2e18 (2023-08-07): [Factory Reset] Fix keeping settings (#4263)

Fixes: #4263
Fixes: #2746
Fixes: #2226
closed 2023-08-07 by TD-er