letscontrolit/ESPEasy #2228: Nextion Display dont work since Release mega-20190106 (HW-Serial)

schamu, 2019-01-09:
### Checklist
- [ x] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2019-01-09:
Can you give some more information on what is working, what not, what serial pins you use, etc.
The only thing you did is writing a subject, which is  indeed short, descriptive and to the point.
But please give a bit more information on what is happening.

schamu, 2019-01-09:
I'm sorry, my English is so bad that I now use the Translator.
Since version 20190106 I can not communicate with the Nextion. I use the GPIO 13 and GPIO 15 with 115200 baud. With version 20181231, everything still worked. After updating to 20190106, the Nextion did not work, either sending text to the display or sending buttons from the display to the controller. With the downgrade to 20181231 everything works again.

MadNBG, 2019-01-18:
Isn't Nextion not on 9600 baud? I am not sure..

schamu, 2019-01-18:
The display can also be set to 115200 baud. I did that. Up to version 20181231 it worked well.

schamu, 2019-02-06:
Thank you, but the mistake is still there.

TD-er, 2019-02-06:
I was afraid so.
Today I am also testing with the Eastron SDM energy monitor and that one also isn't working now on HW serial so I expected something like this.

TD-er, 2019-02-11:
I fixed some minor things.
At least it fixed some strange issues with the Eastron energy monitor.
Could you test this [test build](https://www.dropbox.com/s/jaqfoqxv7jj50xt/ESPEasy_mega-20190202-69-PR_2235.zip?dl=0) to see if it also fixes the issues with the Nextion?

schamu, 2019-02-11:
I'm sorry, it still does not work.

TD-er, 2019-02-11:
Thanks for testing, I will revert the Nextion changes for now and continue to look for a solution.

TD-er, 2019-02-13:
I just made a new [test build](https://www.dropbox.com/s/g5qo3h2rggylqf8/ESPEasy_mega-20190212-73-PR_2235.zip?dl=0) in which I changed a lot regarding the HW/SW serial wrapper.
It now works with the Eastron plugin for both SWserial and HW serial and SWserial is now using the old library we used until 20180131.
So hopefully it is also working again with the Nextion plugin

TD-er, 2019-02-14:
N.B. the PR has been merged this morning, so the next nightly build will be (almost) the same as the test build I linked in the previous post.

schamu, 2019-02-14:
Hello, the prerelease works. Thank you.

schamu, 2019-02-16:
It is not quite so. Apparently, the assignment of HW0 and HW0 swap is not correct. If I rebuild the device nextion it will work with HW0 swap, after a reboot it will not work anymore. If I then switch to HW0 it is then again until I do a reboot.

TD-er, 2019-02-16:
OK, that's for sure a bug.
I will look into it, but it is a bit strange another plugin using the serial swap option (GPS) is still working after reboot.
So it may be an issue in the Nextion plugin itself.

TD-er, 2019-02-25:
Just to make sure I will not put a lot of effort in searching for something that is already fixed...
Can you test the last build to see if the Nextion is still working after a reboot?
And what do you have to do to make it work again? Just pressing 'submit' on the device page is enough?

schamu, 2019-02-25:
It looks like it works now.

TD-er, 2019-02-25:
I have labelled it as being fixed.
If it still appears to be broken, please re-open this issue. (or open a new one and refer to this one)
Thanks for testing.

closed 2019-02-25 by TD-er