letscontrolit/ESPEasy #2233: Puya image - where to find or what to use

sanderlv, 2019-01-11:
ESP-01S to be flashed with puya image. Read a lot, but what image to be used? In the releases there is no image name with "puya" in it?

https://github.com/letscontrolit/ESPEasy/releases

sanderlv, 2019-01-11:
Thus, downloading the package and using the "ESP_Easy_mega-20190108_normal_ESP8266_1024.bin" firmware should work? It doesn't :-(

TD-er, 2019-01-11:
What is working and what not?
And are you sure it is an 1 MB flash node? There are also ESP-01 versions with less.

Can you output the log from the serial port?

sanderlv, 2019-01-14:
Hi, I can flash the firmware via the tool succesfully. But no serial response afterwards. I have tried several images (1mb) etc. But nothing works. 1 Image I have found via a forum that worked (puya build) but I accidentally deleted it and I cannot find it anymore.

TD-er, 2019-01-14:
Do you have any idea on how old this working image could have been? (estimate up-to 2 months is close enough)

TD-er, 2019-01-14:
When you connect GPIO0 to GND, you will force the unit to boot in flash mode.
So as long as that connection exists, it will never boot to load the flashed firmware.

sanderlv, 2019-01-14:
Yes, and not working with or without...

TD-er, 2019-01-14:
Have you also tried to explicit pull it to high via a resistor? (4k7 Ohm for example, between 3v3 and GPIO0)

petefirefox, 2019-01-19:
I have quite a lot of experiments recently with ESP8266-01 modules that have PUYA flash on board.
I also face quite some issues flashing them.
The symptoms are that I can flash them the 1st time fine, but if I want to reflash them (as I do some experiments with a custom plugin that I build), then I do get totally unreliable results.

But I found a workaround. Maybe that workaround can also be helpful here and also give some indication on a potential fix/workaround.

If I do flash the following image:
ESPEasy_R147_RC8/ESPEasy_R147_1024.bin 

And then I have to reset and check in the serial monitor that all the sectors get erased.
Once the serial monitor shows something with "deepsleep", then I can again set the device into flash mode.
If I now flash the new image based on 2.50 core, then I get a reliable behaviour.

I was also trying to flash instead of the R147_RC8 image the beta 2.50 image as of 31122018. But that also resulted in none-stable results.

Effect is that actually the old image/configuration is not or only partially overwritten by the new binary.
If that is actually intended, then I have to apologize for my ignorance and misunderstanding.

Could you try if that workaround works for you, too?

TD-er, 2019-01-19:
The zip file with nightly builds also has "empty" files, which you can flash. These contain all 0xFF and thus effectively erase all sectors of the flash chip.
That may be easier to do as an intermediate step between flashing versions.

petefirefox, 2019-01-19:
I actually tried those. But it also did not work reliable.
For me the only reliable method of clearing the ESP8266-01 and getting new binary on it at the moment is as described above.
I probably have done somewhere around 50 flashing rounds on the same device with the last week. And the above method worked fine every time. The other trials mostly failed.

If someone is only flashing a new ESP8266-01 a single time, then the issue might not pop up. But if you do multiple flashes, then you might run into it. At least I did heavily on multiple parts, as I was taking a new ESP8266-01 if the old device was unusable at some point in time.

Muller59, 2019-05-06:
Using the esptool.py like this
      c:\python27\python esptool.py --port COM4 --baud 115200 erase_flash 
always works for me in case of failed flashes
download from here -> https://github.com/espressif/esptool

closed 2019-08-26 by TD-er