letscontrolit/ESPEasy #2236: [Plugins] P044_P1WifiGateway. shouldn't this be freed at some point in PLUGIN_EXIT?

Barracuda09, 2019-01-14:
See link:

https://github.com/letscontrolit/ESPEasy/blob/0977b7e70e7771d04ea4eea11734e024b27f222f/src/_P044_P1WifiGateway.ino#L138

And maybe change this to new and delete?

TD-er, 2019-01-14:
As we speak, I am updating this plugin to use the `Plugin_task_data` and indeed it needs to free the data.
I was even thinking about using a std::vector for this one.

TD-er, 2019-02-25:
The code for the `P044_P1WifiGateway` is reverted to some state before my fixes of this issue.
So I added a quick fix for this memory leak.
Later we will have a proper look at the code to see what was causing the issues with the ser2net plugin.

Anyway this issue is now fixed, so I will close it.

closed 2019-02-25 by TD-er