letscontrolit/ESPEasy #2242: mega-20190106 load causes ESP32 load to SW reset

cmorrow101, 2019-01-17:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

cmorrow101, 2019-01-17:
ESP32
When switching tabs within the GUI the uP initiates a software reset before the webpage is able to completely load.
Confirmed load mega-20181231 operates normally.
Steps
load unit webpage Main tab, switch to config tab and unit resets. 

Config:
DOIT ESP32
Version mega-20190106 and higher

TD-er, 2019-01-17:
I've also seen it happen on my test node, also with Platformio 1.6.0 (released a few days ago).
Just opening the advanced page leads to crashes on my test node running ESP32.

Good to know the last build of 2018 is still working.

TD-er, 2019-02-02:
Apparently PR #2275 may fix some occurrences of this issue, but not all.
See https://github.com/letscontrolit/ESPEasy/commit/bdda128be82798632800413a5716f582db3a07aa

tonhuisman, 2022-04-18:
This has been resolved some time ago, so can be closed.

commit 0ad7fad5 (2019-02-01): Merge pull request #2275 from joniw/fix-crash-esp32

fix reset bug for esp32 by casting explicitly, should fix #2242
commit bdda128b (2019-02-01): fix reset bug for esp32 by casting explicitly, should fix #2242
closed 2022-04-18 by TD-er