letscontrolit/ESPEasy #2243: ESP8266 ESP-08S Flash

jannnfe, 2019-01-17:
Hello,
i want to flash a ESP8266 ESP-08S with ESPEasy.
I connected ESP -> FTDI:

GND -> GND
VCC -> VCC
GPIO 0 -> GND
GPIO 2 -> VCC
GPIO 15 -> GND
RX -> TX
TX -> RX

I was able to successfully flash the firmware with the esptool:
````esptool.py --port /dev/tty.usbserial-00000000 write_flash 0x00000 ESP_Easy_mega-20190116_test_ESP8266_4096_VCC.bin 
esptool.py v2.2
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 901136 bytes to 579239...
Wrote 901136 bytes (579239 compressed) at 0x00000000 in 51.4 seconds (effective 140.2 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting...
````

The ESP-08S then creates a AP i can connect to and set wifi settings. It reboots and i can edit the main settings. So far so good. i disconnect GPIO 0 and GPIO 2 and GPIO 15 still connect. When i power off the ESP and power on i will get this serial log:

````
ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
⸮U90 : 

INIT : Booting version: mega-20190116 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
91 : INIT : Warm boot #58 - Restart Reason: Exception
94 : FS   : Mounting...
119 : FS   : Mount successful, used 76053 bytes of 957314
508 : CRC  : program checksum       ...OK
546 : CRC  : SecuritySettings CRC   ...OK 
654 : INIT : Free RAM:23456
655 : INIT : I2C
656 : INIT : SPI not enabled
672 : INFO : Plugins: 74 [Normal] [Testing] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
675 : WIFI : Set WiFi to STA
708 : WIFI : Connecting CENSUR attempt #0
709 : IP   : Static IP : 192.168.178.36 GW: 192.168.178.1 SN: 255.255.255.0 DNS: 192.168.178.1

Exception (0):
epc1=0x402024a1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000003c depc=0x00000000

ctx: cont 
sp: 3ffff310 end: 3fffffd0 offset: 01a0
````

The ESP reboots and reboots. Did I wire something wrong? what am I doing wrong?
The ESP-08S is similar to that ESP-07. Can somebody help me please?

Thanks!

jannnfe, 2019-01-17:
Hey @TD-er,
Thank you for your answer. I've only worked with a Wemos D1 Mini so far, where everything went out of the box. I wanted to try something new, so I'm not so familiar with pullup and pulldown yet.

I found the following in the Datasheet of the ESP-08S:
```
Mode         | GPIO15   | GPIO0   | GPIO2
   --------- | -------- | ------  | --
UART         |    low   |   low   | high
Flash Boot   |   low    |  high   | high
```

Here is the Datasheet: https://ecksteinimg.de/Datasheet/Ai-thinker%20ESP-08SWIFI%20Module-EN.pdf

Can you please explain what i have to do to get ESPEasy running.
Thank you :)

TD-er, 2019-01-17:
Also if you solder it yourself, make sure to add some capacitors.
Add a 100 nF capacitor as close as possible to the VCC and GND.
And parallel to that one a 10 uF capacitor, but that one can be a bit further away if the space is tight.
You may also want to add somewhere a 470 uF capacitor to stabilize the power supply.

jannnfe, 2019-01-18:
Thank you! Now its working!
But i now have a other Problem with Deep sleep.
Want to wake up device from deep sleep with button pressed. I know Connecting Button to RST will reboot device and wake up device, but i want to detect double pressed and long pressed Button on Switch Input device. 
Do you have any idea?
Thanks! :)

TD-er, 2019-01-18:
I guess if you want to do that, you will need some other circuit in front of the ESP.
One to perform some kind of counting and also send a pulse for the first keypress or something.
If connected directly to the RST pin, then it will simply reset the ESP when pressed.

tonhuisman, 2022-05-07:
This is unanswered, but not sure if it is still relevant

closed 2022-05-07 by TD-er