letscontrolit/ESPEasy #2247: ESP flasher 0.04.002 does not find bin folder

TungstenE2, 2019-01-18:
ESP flasher 0.04.002 does not find bin folder

[image]

Grovkillen, 2019-01-18:
Sorry for that. What folder path are you using?

Will fix

micropet, 2019-01-18:
Hi @Grovkillen.

For me, he dont find my Wemos on COM9.

Grovkillen, 2019-01-18:
If you want a minute or so? Still don't find it?

TungstenE2, 2019-01-18:
@Grovkillen before I only unpack the zip, start the exe and I can select the bin from drop down. With this version it is not working.

TungstenE2, 2019-01-18:
damn, I did not want to close... how to open again?

TungstenE2, 2019-01-18:
ah... ;-)

TD-er, 2019-01-18:
The flasher will seek for a directory named `bin` in the directory where the executable is running.
If it can't find it (since you're running it from some temp directory when running directly from the ZIP), I can imagine the selection list of found .bin files remains empty.

micropet, 2019-01-18:
@Grovkillen,

COM9 will not be found even after 10 minutes. Even a Refreh does not help.
Putty has no Problems with COM9

Grovkillen, 2019-01-18:
Ok, I will look into it.

TungstenE2, 2019-01-18:
sorry, if I was not clear. zip was unpacked to local folder. exe was started from locak folder. not tmp directly from zip.

Grovkillen, 2019-01-22:
> Hello
>
> I have a same issue. Last working version is: 0.02.005 (20181203 release)
> after this release no bin file in the dropdown.
> I extracted the zip before. ;-)
>
> Thx
> T
>
> [image: image]
> <https://user-images.githubusercontent.com/34792633/51553127-04f45980-1e6a-11e9-821b-a10653d93322.png>
>
> [image: image]
> <https://user-images.githubusercontent.com/34792633/51553177-1c334700-1e6a-11e9-831a-274afae9df11.png>
>
> [image: image]
> <https://user-images.githubusercontent.com/34792633/51553499-ea6eb000-1e6a-11e9-8317-c378e9e98459.png>

Please give me the folder path so I can see if that is causing the error.

TungstenE2, 2019-01-22:
C:\Users\Username\Desktop\ESP Easy\ESPEasy_mega-20190116\bin

Grovkillen, 2019-01-22:
Thanks! Will try to get time tonight or tomorrow morning.

TD-er, 2019-01-22:
Did you also change something regarding the detection of the COM port?
Yesterday I couldn't find the com port of a newly connected device, while PlatformIO could find it.

Grovkillen, 2019-01-22:
Yes I did "fix" some things in regards of com detecting. Will dig into it
tomorrow.

manhal99-hash, 2022-02-20:
Hello there, 
Me too, I have this Issue but when I wont to find the ZIP file I could not open it ! Why ?

tonhuisman, 2022-02-20:
> Hello there, Me too, I have this Issue but when I wont to find the ZIP file I could not open it ! Why ?

The ESPEasy Flasher needs to have a .bin file, and won't read that from a .zip file automagically. The .bin files need to be in the bin subfolder in the folder that the ESPEasy flasher is started from.

tonhuisman, 2023-09-01:
The ESP Easy Flasher has been removed from the distribution so this issue can be closed.

closed 2023-09-01 by TD-er