letscontrolit/ESPEasy #2248: Set INA219 into Sleep, while Deep Sleep of the ESP

Alpenjohn, 2019-01-18:
I am using a few ESP8266-12E and ESP8266-12F Modules to measure Temperature, Humidity, Light and so on, only powered by one 18650 Li-Ion Battery.

Mostly, i am using the BME280 as sensor.
I am sending data every 10 Minutes and then set the ESP into deep sleep.
To measure the voltage of the battery, i am using an INA219.

The Battery life is about 5-6 weeks, if the battery is relatively new (LG-HG2 with about 2600 mAh).
This is a very poor lifetime for such a huge battery and the deep sleep enabled.

In Summary, i am using the following:
1 HT73333 with 8 µA Quiescent Current
1 BME280 with about 4 µA Quiescent Current
1 ESP8266-12E with about 20? µA Quiescent Current
1 INA219 with about 1 mA while not in Sleep Mode.

I measured some configurations and get about 70 to 120 µA with a connected INA219. Now i am relatively sure, the INA219 does not go into sleep mode, while the ESP8266 is in deep sleep, which drains my batteries.

Is it possible to use the INA219 sleep mode, while the ESP8266 is in deep sleep?
This will help a lot and should safe a lot of energy.
And it does not really make sense to use an battery powered project and don't know how much capacity is left in the Battery.

TD-er, 2019-01-18:
That sounds like a valid feature request :)
I know the BME280 is actively been put to sleep mode, so indeed why not the INA219.
Why not use the A0 input on the ESP by the way?
I know you will need a voltage divider to do so and I can imagine those resistors may also take quite a big quiescent current themselves.

Wiki591, 2019-01-24:
Yes, using the INA219 sleep mode would be a real enhancement extend battery lifetime.
I am using a quite similar configuration, but feeding the INA219 from one pin of the ESP as a workaround to solve this problem. Due to the fact that the pins of the ESP may deliver 12mA max and the INA pulls 1mA max it works perfectly for me but occupies one pin.
Btw.: working with a voltage devider of 100k/400k to get the needed 5V measurement range for the 1V A0 (max. 4.2V, min. 3.2V) pulls somewhat of 7.5 µA quiescent current - following Ohm's law.

PR #5089 (2024-07-27): [P027] Add optional Powerdown mode
Resolves #2248 
Resolves #3481 

Features:
- P027 Add Powerdown mode (optional) to reduce the quiescent current when not taking measurements, f.e. during sleep mode in battery operated configurations.
  - This feature is not available in `LIMIT_BUILD_SIZE` builds (Display, Collection, Climate) for bin size reasons.
- Extend/Update documentation.
- P027 Fix sending single values via Controller as reported in #3481

TODO:
- [ ] Testing
- [ ] Update documentation
closed 2024-07-27 by TD-er