letscontrolit/ESPEasy #2249: Email notifications do not work

gumileeev, 2019-01-18:
Hello.
ESP_Easy_mega-20190116_normal_ESP8266_4096 nodemcu v3
Email notifications do not work.

TD-er, 2019-01-18:
How do you try to send it?
Did the same configuration work before?

gumileeev, 2019-01-18:
> How do you try to send it?
> Did the same configuration work before?

I have not used this feature before. E-mail settings are 100% working, I tried 2 mails, with these parameters, I successfully send letters "Domotics". It does not work at the very beginning, when I try to send a test from the add email menu.

TD-er, 2019-01-18:
Can you show the settings (with account login + password removed)?

gumileeev, 2019-01-19:
Of course [screen](http://prntscr.com/m94xwl)

TD-er, 2019-01-19:
Just as a test, could you try to use only "Latin" characters?
I don't know how well Unicode characters work.

gumileeev, 2019-01-19:
Thank! Mail really earned! The certificate SSL was to blame

closed 2019-01-29 by gumileeev