letscontrolit/ESPEasy #2250: DHT22 Checksum Error

gumileeev, 2019-01-18:
Hello.
ESP_Easy_mega-20190116_normal_ESP8266_4096 nodemcu v3
When using the DHT22 humidity sensor (AM2302) I get problems with data collection - I get the error "DHT: Checksum Error", the update time is 10 seconds, I get data only every 5-10 times. I get data through time. On the assembly that I downloaded a month ago (I don’t remember the version), everything worked fine. I tried another humidity sensor, and putting the firmware on another board did not help.

TD-er, 2019-01-18:
Not much has changed to this plugin the last few months.
See its [history (P005)](https://github.com/letscontrolit/ESPEasy/commits/mega/src/_P005_DHT.ino)

gumileeev, 2019-01-18:
> Not much has changed to this plugin the last few months.
> See its [history (P005)](https://github.com/letscontrolit/ESPEasy/commits/mega/src/_P005_DHT.ino)

I can not understand, I can fix it on my own? Receives data, but through time. Or should I wait for the update, and the problem will go away?

TD-er, 2019-01-18:
Well you stated you used a version of about a month ago, and the last big change to that plugin was merged on September 29.
So right now I don't have a clue what could be wrong here.

Are there other plugins active?

TD-er, 2019-01-19:
I just installed a Sonoff TH16 here with a Sonoff AM2301 sensor mounted to it.
It is reading just fine here.
What setting are you using for the DHT type?
[image]

And what version of the AM2301 are you using?
If I remember correctly, the Sonoff version is not really connected how the sensor itself should normally be used.

gumileeev, 2019-01-19:
I use the plugin version - which comes with the build, unfortunately I don’t know how to install another plugin inside the build. I use DHT22 (AM2302). Here is a [screen](http://prntscr.com/m95dj2) with settings. There is no other way to use another gpio.

gumileeev, 2019-01-19:
I attach a [log](http://prntscr.com/m95exv)

uzi18, 2019-09-04:
@gumileeev you can try latest builds

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er