letscontrolit/ESPEasy #2251: Upgrade Sonoff Pow R1 from Mega-20181220 to Mega 20190116 fails

2bontb, 2019-01-19:
### Summarize of the problem/feature request

Upgrade Sonoff Pow R1 from Mega-20181220 to Mega 20190116 fails.
Opening the device page Energy (AC) - HLW8012 (POW r1) [TESTING] on the new release reboots the Sonoff. 
Tried the following to see if it solves the problem:
- On the factory reset page on the old release select a R1, and do factory reset.  On new release it is stillnot possible to add the energy reading.
- On new release try fa tory reset.

I use distributed version from bin directory:  hard_sonoff_POW_4M

### Expected behavior

Upgrade should be succesful.
Or creating clean config on 2019016 should be successfull.

### Actual behavior

After factory reset:
Opening Energy (AC) - HLW8012/BL0937 [TESTING] on new release reboots the Sonoff.

### Steps to reproduce

Take a Sonoff R1, and create an energy reading device. This is not possible. 

Dit not try this; It is build in. I have to disconnect it fits.

### System configuration

Hardware:
Sonoff R1

ESP Easy settings/screenshots: 
NA

### Rules or log data

NA

TD-er, 2019-01-20:
I just updated one of my POW r1 units to the latest build (20190116)

ESP Easy | Information |
 -----|-----|
Build:⋄|20103  - Mega|
Libraries:⋄|ESP82xx Core 2.5.0-beta2, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support|
GIT version:⋄|mega-20190116|
Plugins:⋄|7 [Normal] [Sonoff POW R1/R2]|
Build time:⋄|Jan 16 2019 03:23:22|
Binary filename:⋄|ESP_Easy_mega-20190116_hard_core_250_beta_SONOFF_POW_4M.bin|

The upgrade is just working fine.

Using these settings:
[image]

Also deleting the sensor and adding again works, but there is a recent change in this plugin.
After adding, you first have to select the predefined Pin Settings and set it to the right value.
Then press submit and add a name + check enabled.
Then it should work.

2bontb, 2019-01-21:
I retried a clean install
1. Take it from power, and flashed it with 4M_blank.
2. Flashed it with mega-20190116_hard_core_250_beta_SONOFF_POW_4M.bin
3. Set the firmware reset options (See settings below)
4. Go to the device page. 
5. Correcting switch settings and system info
6. Add device for the HLW8012

Still the same problems.
let me share some pages and setup data what I have and see.

Before the upgrade my system looks like:
[image]

[image]

[image]

Aftger the upgrade, the pages look like:
[image]

[image]

You see this last page draws always the same, but is incomplete. After showing this, the system reboots.
So I cannot execute your proposed actions.
I always did so far set the factory reset options as displayed above on the old/new release, before going to the device page.

Any suggestions of things to try?

2bontb, 2019-01-21:
The Json file of the config:

  |  
-- | --
System |  
Build | 20103
Git Build | "mega-20181220"
System   libraries | "ESP82xx   Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support"
Plugins | 7
Plugin   description | "   [Normal] [Sonoff POW R1/R2]"
Local time | "1970-00-00   00:00:00"
Unit | 123
Name | "SonoffWerkkamer"
Uptime | 1466
Last boot   cause | "Cold   boot"
Reset   Reason | "Power   on"
Load | 18.5
Load LC | 6944
Free RAM | 18128
WiFi |  
Hostname | "SonoffWerkkamer-123"
IP config | "DHCP"
IP | "10.2.1.123"
Subnet   Mask | "255.255.255.0"
Gateway IP | "10.2.1.138"
MAC   address | "2C:3A:E8:3B:70:B6"
DNS 1 | "10.2.1.138"
DNS 2 | "0.0.0.0"
SSID | "xxxxx"
BSSID | "7A:8A:20:67:B0:0C"
Channel | 11
Connected   msec | 87944771
Last   Disconnect Reason | 1
Last   Disconnect Reason str | "(1)   Unspecified"
Number   reconnects | 0
RSSI | -57
Sensors |  
0 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "button"
NrDecimals | 0
Value | 0
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "false"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 0
Type | "Switch   input - Switch"
TaskName | "ButtonWerkkamer"
TaskEnabled | "true"
TaskNumber | 1
1 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "relay"
NrDecimals | 0
Value | 1
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "false"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 0
Type | "Switch   input - Switch"
TaskName | "PowerWerkkamer"
TaskEnabled | "true"
TaskNumber | 2
2 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "Voltage"
NrDecimals | 0
Value | 227
1 |  
ValueNumber | 2
Name | "Current"
NrDecimals | 2
Value | 0
2 |  
ValueNumber | 3
Name | "ActivePower"
NrDecimals | 0
Value | 0
3 |  
ValueNumber | 4
Name | "PowerFactor"
NrDecimals | 0
Value | 0
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "true"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 300
Type | "Energy   (AC) - HLW8012 (POW r1) [TESTING]"
TaskName | "EnergieWerkkamer"
TaskEnabled | "true"
TaskNumber | 3
3 |  
TaskValues |  
0 |  
ValueNumber | 1
Name | "Uptime"
NrDecimals | 0
Value | 1465
1 |  
ValueNumber | 2
Name | "RSSI"
NrDecimals | 0
Value | -57
2 |  
ValueNumber | 3
Name | "Load"
NrDecimals | 0
Value | 19
3 |  
ValueNumber | 4
Name | ""
NrDecimals | 0
Value | 0
DataAcquisition |  
0 |  
Controller | 1
IDX | 0
Enabled | "true"
1 |  
Controller | 2
IDX | 0
Enabled | "false"
2 |  
Controller | 3
IDX | 0
Enabled | "false"
TaskInterval | 300
Type | "Generic   - System Info"
TaskName | "Uptime"
TaskEnabled | "true"
TaskNumber | 4
TTL | 60000

2bontb, 2019-02-17:
I retried it again today with the latest Load:  

I flashed it with 4M blank,
I flashed it with 20190216 SONOFF 4M
The following is the trace of the serial:
See >>>> where I added the Energy AC device.
Hope you can now find my problem becuase my Sonoff is unusible with newer loads.

RESET: Resetting factory defaults... using default settings
RESET: Warm boot, reset count: 0
RESET: formatting...
RESET: formatting done...
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
⸮⸮⸮U87 : 

 INIT : Booting version: mega-20190216 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
88 : INIT : Warm boot #12 - Restart Reason: Software/System restart
90 : FS   : Mounting...
115 : FS   : Mount successful, used 75802 bytes of 957314
399 : CRC  : program checksum       ...OK
410 : CRC  : SecuritySettings CRC   ...OK 
518 : INIT : Free RAM:30624
519 : INIT : I2C
520 : INIT : SPI not enabled
531 : INFO : Plugins: 7 [Normal] [Sonoff POW R1/R2] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
534 : WIFI : No valid wifi settings
536 : WIFI : Could not connect to AP!
537 : WIFI : Set WiFi to AP
1458 : WIFI : AP Mode ssid will be Sonoff POW R1/R2_0 with address 192.168.4.1
2792 : WD   : Uptime 0 ConnectFailures 0 FreeMem 27472 WiFiStatus 0
>settings

System Info
  IP Address    : 0.0.0.0
  Build         : 20103
  Name          : Sonoff POW R1/R2
  Unit          : 0
  WifiSSID      : ssid
  WifiKey       : wpakey
  WifiSSID2     : 
  WifiKey2      : 
  Free mem      : 26664

Ok
>WifiSSID xxx

Wifi SSID:xxx

Ok
>WifiKey xxx

Wifi Key:xxx

Ok
>Clearaccessblock
Allowed IP range : All Allowed

Ok
>save

Ok
[... 160 lines cut ...]
596 : WIFI : Connecting DeHeijIOT attempt #0
598 : WIFI : Not configured in Station Mode!!: DeHeijIOT
1933 : WD   : Uptime 0 ConnectFailures 0 FreeMem 29008 WiFiStatus 0
2821 : LoopStats: shortestLoop: 1107 longestLoop: 13612 avgLoopDuration: 1159.42 loopCounterMax: 27100 loopCounterLast: 1906 countFindPluginId: 0
2822 : Scheduler stats: (called/tasks/max_length/idle%) 1905/138/6/73.20
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 11
cnt 

connected with DeHeijIOT, channel 11
dhcp client start...
ip:10.2.1.123,mask:255.255.255.0,gw:10.2.1.138
4464 : WIFI : Connected! AP: DeHeijIOT (7A:8A:20:9F:5F:B3) Ch: 11 Duration: 3766 ms
4465 : WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_GOT_IP
4466 : WIFI : DHCP IP: 10.2.1.123 (Sonoff-POW-R1/R2-0) GW: 10.2.1.138 SN: 255.255.255.0   duration: 21 ms
4471 : Webserver: start
4472 : WIFI  : Arduino wifi status: WL_CONNECTED ESPeasy internal wifi status: ESPEASY_WIFI_SERVICES_INITIALIZED
pm open,type:2 0
31932 : WD   : Uptime 1 ConnectFailures 0 FreeMem 24472 WiFiStatus 3
32821 : LoopStats: shortestLoop: 115 longestLoop: 885930 avgLoopDuration: 126.37 loopCounterMax: 260869 loopCounterLast: 222312 countFindPluginId: 0
32822 : Scheduler stats: (called/tasks/max_length/idle%) 222311/2058/6/97.20

TD-er, 2019-02-17:
I like to know then what's different here.
I just uploaded to a POW r1:
Build:⋄
20103  - Mega
Libraries:⋄
ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
GIT version:⋄
mega-20190216
Plugins:⋄
7 [Normal] [Sonoff POW R1/R2]
Build time:⋄
Feb 16 2019 03:26:21
Binary filename:⋄
ESP_Easy_mega-20190216_hard_SONOFF_POW_4M.bin

[image]
Did you also test the node (after flashing) while connected to mains power?

2bontb, 2019-02-18:
No I never connected it back to mains after the flash. Just becuase I could never configure the Devices. What do you suggest; Load back the old release and backup and upgrade this one. [however I did this already when I did OTA upgrade and lost the device]. Or flash it with 4M clean, after this, flash it with 20190216. And that just connect it too mains? How do you advise to set the Wifi? I did set this first via the serial as you can see in the log above. Or should I connect and configure it to the exposed AP?

2bontb, 2019-02-18:
Other question: how can I decode the stack? Does this give some clue?

2bontb, 2019-02-18:
I just did the following:
Flashed 4M Blank
Flashed 20190216_hard_SONOFF_4M
accessed the AP, and did set the WiFi
Connected it to Mains
Tried to add on slot 1 the HLW802. Same result. Page immediate dies.
In the weekend I will try one of my two other POWR1. See If they have same problem.
For now I flash this one back to 20181220.

TD-er, 2019-02-18:
Stack decoding is something I also struggle with.
The binaries are not compiled using the Arduino IDE, but using PlatformIO, so I am not sure if decoding the stack will result in something useful.

And if the December version is working, it is best to return to that one for now.
Not sure what's going on here. I will try to reset mine to see what is wrong here.
Maybe the extension to add more models is breaking things? 
I thought I had made some more elaborate description of how to setup this sensor, in the [documentation](https://espeasy.readthedocs.io/en/latest/Plugin/P076.html#p076-page) but apparently not.

In my memory, you had to setup all parameters first and then enable it.
But even then it should not crash.

2bontb, 2019-02-19:
TD-er, the docu is clear. But I can get in no way the page that has the setting. As soon as I open the page the reboot occurs. I will try oin weekend twothings
1) Try it on other sonoff I have (But that one is build i, so takes some effort to get toit)
2) Build the source via IDE, so perhaps we can decode the stack after a crash. I assume IDE builds are still possible. I did this in the past, but couple of months ago I did step over to PlatformIO. (I have some sensors using experimental code [Itho])

TD-er, 2019-02-19:
Hmm "good news".
I am able to reproduce it on my node.
So now I can try and see what's going on.

2bontb, 2019-02-19:
Thats indeed good news. 
I was just compilinjg the source on IDE using core 2.4.2.
But I had to fix already two compile errors. Looks like PlatformIO does not bother on these.
1. Had to move  #include <ESPeasySerial.h> to ESPeasy-Globals.h
2. Had to move two functions void ICACHE_RAM_ATTR p076_hlw8012_cf1_interrupt() and void ICACHE_RAM_ATTR p076_hlw8012_cf_interrupt() in _PL076_HLW....ino above the calling functions, becase the were not yet defined.
Let me know if I still need to load this IDE version to see if I can get a core dump analyses.

TD-er, 2019-02-24:
Should be fixed in PR #2348
In short, it was working when you set all parameters yourself and kept the selector to "custom", but as soon as you would try to save the settings with the pre-defined config set to some other value, it would crash.
This was due to something I changed, but even then it took me a while to find it.

TD-er, 2019-02-25:
Close this one, since there's a new one: #2355

closed 2019-02-25 by TD-er