letscontrolit/ESPEasy #2254: mega-20190116 causes missingh mhz19 co2values

pwassink, 2019-01-20:
version mega-20190116 

After updating to mega-20190116 i have several different esp/types of nodes who stop sending the C02 values from the MHZ19 co2 sensor, wifi connection is still there, other sensors on same esp still work fine. data destination is Domoticz, data does not arrive there so it must be a local (esp) problem.
I see the same symptoms on 4 different nodes here.

Logfile content shows this:  
MHZ19: Error, timeout while trying to read
MHZ19: Unknown response: 0 0 0 0 0 0 0 0 0

Anyone with same behaviour ?

TD-er, 2019-01-20:
Looks to be related to the change in serial I made recently.

What gpio pins do you use?

pwassink, 2019-01-20:
It might be,  lets see, I guess you have found a point Gijs :-)

esp01 = Gpio0 Gpio2 (no problems seen yet)
esp02 = Gpio14 Gpio12
esp08 = Gpio14 Gpio12
esp18 = Gpio14 Gpio12

Peter

TD-er, 2019-01-20:
I have two nodes here I can also test later this afternoon

pwassink, 2019-01-20:
Ok, 
Fault is not immediate, they stop sending after a while, and are quite persisting in that then, reboot or sw downgrade is not enough toie get them sending data again. They need a powercycle and some time between off and on.

TD-er, 2019-01-20:
How long is "after a while" ?
I have just setup a node using one of these sensors and also a BMP280 is connected.
It is using GPIO-14 & -12 for the CO2 sensor.

pwassink, 2019-01-20:
a couple of hours, yesterday everything was working, this morning i had 3 out of 4 with same status.

TD-er, 2019-01-20:
And they all booted around the same time, before they all stopped working?

pwassink, 2019-01-20:
yes, all were restarted/updated within an hour i think, was updating them yesterday evening. and during the night they stopped working

TD-er, 2019-01-21:
If any missing updates are not a big deal, I would suggest the latest version.
To keep on the version before the serial port wrapper, you could use 20181231. (I think I committed the changes this year)

pwassink, 2019-01-21:
i did downgrade the 3 "problem" esp's to 20181231 yesterday, can confirm they are still sending data with no hardware changes, so those three esp's still using gpio12 and gpio14 in my case

Any luck  with recreating it Gijs ?

TD-er, 2019-01-21:
Nope, my node is still sending values and it is running the build from January 16th. (core 2.5.0)

pwassink, 2019-01-21:
i used the ESP_Easy_mega-20190116_normal_core_241_ESP8266_4096.bin on those nodes, that has the 2.4.1 core i believe ?

TD-er, 2019-01-21:
@pwassink That's correct.
You can also see the core build in the sysinfo page.

pwassink, 2019-01-21:
I did install that version ESP_Easy_mega-20190116_normal_core_241_ESP8266_4096.bin back to the esp02 here, lets see if it happens again

pwassink, 2019-01-24:
Hmm, 
Maybe the day of the week or the position of the moon has a certain influence, 
I did not see the fault reoccurring here within the past days.

TD-er, 2019-01-24:
It seems to be working fine here too.
It was just about full moon when you reported it, so I guess that should have been it ;)

pwassink, 2019-01-26:
Update, 

i left esp02 running with the ESP_Easy_mega-20190116_normal_core_241_ESP8266_4096.bin version that is still running as it should.

Yesterday i did update the rest of my nodes to ESP_Easy_mega-20190121_normal_core_241_ESP8266_4096.bin

And esp-18 went bozo again at 06:55 local time, all is functioning as it should,  except the MHZ19 Co2 sensor, in the devices tab it is displaying the "last-known-good value" from approx 07:00 and again with the same entries in the Logfile :
MHZ19: Error, timeout while trying to read
MHZ19: Unknown response: 0 0 0 0 0 0 0 0 0

Esp-18 is using Gpio 14 / 12

TD-er, 2019-01-26:
Does it need to do something at that moment? Can be something which could lead to a hickup of the wifi on the ESP?

pwassink, 2019-01-26:
Nothing at all, 

There are no scheduled resets, reboots, backups, wfi resets, dhcp cleanups or any external reason at that time

pwassink, 2019-01-28:
Again 2 hickups today, esp02 and esp18 both went wrong again same faults in the log 
Software 2 versions involved, 0116 and 0121 both 2.4.1 core and 4 Mb version
hardware of esp02 is a nodemcu-d1-mini / esp18 is a nodemcu-v2

TD-er, 2019-01-29:
Can you try this build on one of your nodes?

ESP_Easy_mega-20181109_normal_ESP8266_4096.bin

It is running on one of my nodes with a MH-Z19 and has now an uptime of 54 days 23 hours 21 minutes
That's since a power outage...
This one is running fine with regular updates of the CO2 value.

pwassink, 2019-01-29:
i will, 

but have i already mentioned before that up till version mega 20181231 core 2.4.1 normal  4Mb none of the reported stability issues have been seen, so why this specific version ? 

i will download that specific version en install it to all of my co2-measuring nodes, but the problems appeared in the mega-2019* range not earlier in my humble opinion Gijs ..

TD-er, 2019-01-29:
OK, then it is of no use indeed.

That specific version is running on a version I have which appears to be (sadly unusual) stable.
And you're right, if it was running fine until 20181231, then it is not really worth to try the older ones.

pwassink, 2019-01-29:
Nevertheless, 
Esp02 and esp18 are running on ESP_Easy_mega-20181109_normal_ESP8266_4096.bin now
Esp01 and Esp08 are running on ESp-Easy_mega_20190121_normal_core_241_ESP8266_4096.bin

And we will see

TD-er, 2019-01-30:
A checksum error should be handled more gracefully.
Maybe we should add some threshold on when to start shifting to search for a new good start.
Or a better algorithm to get in sync again.
As far as I know, there has been no change in the code for this in over a year. There has been only a change in the way the serial port connection is being created, so I really don't know why it leads to these issues.

pwassink, 2019-01-30:
I have the idea that the mhz19 sensor itself enters a "blocked" state after a couple of hours miscommunication or as the result of not beeing able to get data out. ?
Motivation: a reboot or even a full sw-update does not solve that status, it has  to be powerless for a minute or so to regain life, i found this behaviour when it is stuck in the status that looks like: 
MHZ19: Error, timeout while trying to read
MHZ19: Unknown response: 0 0 0 0 0 0 0 0 0

The fault esp08 was this morning was still solvable with a remote reboot option from the web console so it was not the same as the complete lockup after several hours i've found and reported a couple of times.

TD-er, 2019-01-31:
It sounds quite strange, since the plugin does send a command to gather new sensor data  to the MH-Z19
So I don't get it why even a soft-reboot is not working.
I will add also some stats to see how often a checksum error occurs (receiving end)
If that happens a lot, it may also happen when sending data to the sensor and maybe the sensor is then put into some unknown command mode.
Maybe some pull-up resistor configuration has changed when I changed the way how serial pins are configured???

pwassink, 2019-01-31:
It might,  

will look into it a bit further tomorrow or during the oncoming weekend perhaps i will be able to find something, is that sensor not that widely used that no-one suffers identical behaviour ?

TD-er, 2019-01-31:
Or not used by people running the latest builds

pwassink, 2019-02-03:
Ok, just curious exactly which core-version of the test build and which Gpio's are you using ?

blb4github, 2019-02-03:
ESP_Easy_mega-20190121_test_core_250_beta_ESP8266_4096.bin, GPIO 13 & 12

TD-er, 2019-02-03:
Now that I think of it, there may be another change in the used SWserial library since start of this year.
We used to have our own version of the SWserial library, which I patched to use less iRAM memory.
But now we are using the version included in the core library and for the core 2.5.0 this may be a newer version than before.
Also there may be some change in the use of interrupts on low-speed connections. (9600 baud and lower)
I must look into that too.

It still doesn't explain why the sensor apparently can get so messed-up it needs a power-down to act normal again.

pwassink, 2019-02-03:
Within 3 hours after updating to ESP_Easy_mega-20190202_normal_core_241_ESP8266_4096.bin my esp15 was locked up again, webconsole reboot did not work, powerdown was required again. 
the other one is still working, will add inhere if it freezes the MHZ19 also

TD-er, 2019-02-04:
I just added a commit to do some improvement on the reading.
See https://github.com/TD-er/ESPEasy/commit/3d507bdbb9dc6dd4aee2ce0c7ce6c809ea7dfb7c

Can you build a test version for it, or do you want me to build one?

pwassink, 2019-02-04:
If you could provide me with a bin i will be happy to put it on both still on gpio12/14 running esp's 
and then it is certain the file is the same, no local influences etc ..

TD-er, 2019-02-04:
OK, took some time, but here is the [test build](https://www.dropbox.com/s/om6wf7dy4w6sv0x/ESPEasy_mega-20190202-58-PR_2235.zip?dl=0)

pwassink, 2019-02-04:
Got it, 

Esp08 and esp18 are running on this testversion with the 2.4.1 core now, they both use gpio12/14 

Lets see !

pwassink, 2019-02-04:
Yes , i have them in the console too now!

will leave those 2  and see at certain interval if those counters increase. 
Keep you posted ..

TD-er, 2019-02-04:
Do you also have a mix of MH-Z19 A/B or only the newer B versions?
It should not matter, just curious.

pwassink, 2019-02-04:
> Do you also have a mix of MH-Z19 A/B or only the newer B versions?
> It should not matter, just curious.

B versions, just added that info , detection worked too :-)

pwassink, 2019-02-06:
smaal update, all are still running fine, the ones with your special version esp8 and 18 show increasing conters but still no faults or checksum errors, current values are: 

Esp08: Checksum (pass/fail): | 1795/0
Esp18: Checksum (pass/fail): | 724/0

and the other one that failed quite consistent esp02 is also still running now with the changed Gpio's 
and gas now Gpio-0/Gpio-2 and upgrade it to the 0202 version core 2.4.1

pwassink, 2019-02-07:
Hmm i was happy too early, i guess 

I just did another round of checks and saw Esp02, 08 and 18 an displayed in red on Domoticz 
all three have frozen Co2 now, and both "specials" have no checksum error counter values, they are stiil at 0  
2 of them i cant powercycle now so will reset them tomorrow and will enable more debug-logging and setup a a syslogserver to capture it i guess ?  

tried a webconsole-remote reboot of esp08, got several MHZ19: Unknown response  0 0 0 0 0 0 0 0 0
That is a node with Special-sw , So a reboot does not reinitialise the Mhz19B as far as i can see now.

At 00:55 it started sending C02 values again from Esp08, and the rules based 3 led indicator worked also, beats me .. 

Esp18 also got a webconsole based reboot tonight and did not gain life again.  all counters were left at 0, logging in mode4 only shows MHZ19: Unknown response  0 0 0 0 0 0 0 0 0

Esp02 the modified one using gpio0/gpio2 has been upgraded to : ESP_Easy_mega-20190202-58-PR_2235_normal_core_250_beta_ESP8266_4 now to test that core combination too.

pwassink, 2019-02-07:
Bummer deleted the post here , try to recreate :

Yesterday evening three of my esp's 02, 08 and 18 went red on Domoticz, no Co2 sensor data. 
two of them have the special version core 2.4.1 , and gpio 12/14. A web console reboot did not solve it, and only produced: MHZ19: Unknown response: 0 0 0 0 0 0 0 0 0 but esp18 co2 measuerements gained life back after approx one hour, ta 0:56 it started sending proper values again . 

Esp08 has received a reboot too, and esp02 also (that one has gpio0/2 now same as esp01) both did not recover, end this morning both got a 2 minute power-down , after that both came back to life. 

I changed the software-version on esp02 to ESP_Easy_mega-20190202-58-PR_2235_normal_core_250_beta_ESP8266_4 so we can test that core-version too 

I also activated syslog for esp08 only in the beginning, with settings als below, if you want a specific setting please ask. 
[image]

TD-er, 2019-02-07:
Maybe you can also itemize your configs in the post (can also be next post, no need to edit this one) showing the following:

- Your internal unit nr (for your own reference like "02, 08, 18)
- Build running 
- nr of success/fail (if available)
- kind of failure/issue

Itemized information is (for me) easier to follow compared to descriptive text.
I also reply from my phone sometimes.

pwassink, 2019-02-07:
22:57 esp08 frozen again, counters Checksum (pass/fail): | 1077/2 put syslog aside when found.

pwassink, 2019-02-08:
Gijs, 

I did some analysis on the last syslog of esp08, during the hours that it is running i saw 78 occurances of: 
2019-02-08T05:27:07.581181+01:00 hub08 EspEasy - - - EspEasy: MHZ19: Unknown response: ff 0 0 0 0 0 0 0 0

Automated search using #cat messages-crash-esp08-0625 |grep MHZ19 | grep response | grep 'ff 0 0 0 0 0 0 0 0' | wc -l

and after changing that cmdline to the second unknown-response variant  linux counted 408 times this line : 
2019-02-08T10:44:06.855432+01:00 hub08 EspEasy - - - EspEasy: MHZ19: Unknown response: 0 0 0 0 0 0 0 0 0

The checksum error counters in your custom-debug-version did not get any higher than 2 as far as i saw during the last couple of days.
the first error message (MHZ19: Unknown response: ff 0 0 0 0 0 0 0 0) came a couple of times six or so directly after each other  before it froze up this morning.

TD-er, 2019-02-08:
It looks like the sensor itself is crashing, but I can't think of a reason why it is doing it now.
In our source there is a command for calling reset of the MH-Z19 sensor, but that's not documented in the datasheet.
So I don't know its status. 
Could you try to call that command on a node which is stuck like this?

pwassink, 2019-02-08:
There has to be some change made after the mega 20181231 2.4.1. 4Mb version which has this results on the MHZ19 as far as i found yet. that one is running for weeks without problem even on gpio 12/14.

will try it next time something stops working, at 23:20 found that esp18 was frozen, mhzreset did not chage that, see above log.

TD-er, 2019-02-11:
That command does seem to do something for the MH-Z19 A version.

```
127136: MHZ19: Sent sensor reset!
137272: MHZ19: Unknown response: ff ff ff ff ff ff ff ff ff
152275: MHZ19: Unknown response: ff 31 f5 4b 42 32 30 30 d
167277: MHZ19: Bootup detected! PPM value: 5000 Temp/S/U values: 23/1/15000.00
197279: MHZ19: Bootup detected! PPM value: 400 Temp/S/U values: 23/1/15000.00
<repeat a few times>
257279: MHZ19: PPM value: 906 Temp/S/U values: 24/64/11554.00
```

So it looks like it cannot be used on the B-version.

pwassink, 2019-02-11:
I could assume something like that command is also used someway in the startup / init of the plugin ?
which might explain why a reboot / reset without power cycle does not solve the problem.

Did not see failures until this afternoon:
15:43 esp08 frozen again, counters Checksum (pass/fail): | 2316/2 put the systlog aside.

TD-er, 2019-02-11:
And just to be sure, these nodes were running fine on older firmware versions?

pwassink, 2019-02-11:
Yes up till the ESP_Easy_mega-20181231_normal_ESP8266_4096 is was and is still ok, 
They would run for weeks without problems

TD-er, 2019-02-11:
I looked into the recent changes in the SWserial library, since that's the one we're now using.
One of the changes made in there was to no longer use interrupts on TX transfers for 9600 baud.
Can you try [this test build](https://www.dropbox.com/s/18bdv1ck991hu2i/ESPEasy_mega-20190202-82-PR_2235.zip?dl=0) ?

N.B. I also removed the core 2.5.0 builds, since they act strange when serving web pages.

pwassink, 2019-02-12:
Esp08 and Esp18 running ESP_Easy_mega-20190202-82-PR_2235_normal_ESP8266_4096.bin
thos two also do send their data in to a syslog server so the log data will be recorded

The other two will follow tomorrow, one of them might be a Mhz19A, well see

pwassink, 2019-02-13:
05:21 Esp18 frozen, Checksum (pass/fail): | 1460/0, have put syslog aside

TD-er, 2019-02-13:
The core 2.4.1 build was not included in that test build, so I just made one only containing that core version. [core 2.4.1 build of same code as in ESP_Easy_mega-20190202-82-PR_2235_normal_ESP8266_4096.bin](https://www.dropbox.com/s/pougyzwe2vqhjmi/normal_core_241_ESP8266_4096.rar?dl=0)
That one uses an older version of the SoftwareSerial library.
If that's working, then I will change the used SW serial library.

pwassink, 2019-02-13:
Starting upgrade to the special-version: firmware.bin for all 4 nodes now, finished at 12:30

First thing i noticed, Esp08 was frozen, this firmware update restarted the MHZ19B , it came back allive, an the init i mean: sensor is giving me a reasonable C02 Value this seemed much faster too

TD-er, 2019-02-13:
I am running the old SWserial library on a test node too and indeed it is working much better
For example the Eastron energy monitoring was having about 20 - 30% of the received lines corrupted, but this is now running perfectly. (no failed checksums yet)

TD-er, 2019-02-13:
I just made a new [test build](https://www.dropbox.com/s/g5qo3h2rggylqf8/ESPEasy_mega-20190212-73-PR_2235.zip?dl=0) in which I changed a lot regarding the HW/SW serial wrapper.
It now works with the Eastron plugin for both SWserial and HW serial and SWserial is now using the old library we used until 20180131.

pwassink, 2019-02-13:
So this is actually the same as the 0202 version but with the same serial lib as the 20181231 version, i will upgrade them directly .. just a moment

Installed ESP_Easy_mega-20190212-73-PR_2235_normal_ESP8266_4096.bin
on Esp0/02/08/18 now
 
lets see ..

[43 more replies]

closed 2019-10-10 by TD-er