letscontrolit/ESPEasy #2258: ESP32 Several settings not responding

berbergh, 2019-01-20:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

ESP32 with releases as of ESP_Easy_mega-20190106_esp32test_1M8_partition.bin do not respond to mouse clicks in the web control page.
Config, Hardware, Rules and Advanced in Tools are failing to respond.
ESP_Easy_mega-20181231_esp32test_1M8_partition.bin still seems to work fine.
The problem does not go away after a power cycle.

### Expected behavior

You should get to the underlying pages

### Actual behavior

You get a page time-out

### Steps to reproduce

1. Load the releases mentioned
2. click the buttons in the web site of the device
3. 

### System configuration

Hardware: WEMOS ESP32

ESP Easy version: See above

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2019-01-20:
The esp32 crashes on some web pages 
I have not yet an idea what may cause it 
It doesn't seem to be a stack overflow. I already checked it by changing the stack to 32k and it still crashes 
Seems to be some (uncaught) exception???

TD-er, 2019-01-20:
I think it is a duplicate of #2242, so I will close this one.
If it isn't a duplicate, please re-open.

closed 2019-01-20 by TD-er