letscontrolit/ESPEasy #2265: mega-20190121 WiFi password with special characters ie "#," login not possible

tertel, 2019-01-24:
After connecting with ESP_Easy_0 and choosing own WiFi in the setup, I get the following messages on serial monitor:

23 : INFO : Plugins: 46 [Normal] (ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
627 : WIFI : Set WiFi to STA
659 : WIFI : Connecting Asq2!iOpy29d attempt #0
661 : WIFI : Not configured in Station Mode!!: Asq2!iOpy29d
1996 : WD   : Uptime 0 ConnectFailures 0 FreeMem 21696 WiFiStatus 0
3617 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2857 ms
4442 : WIFI : Connecting Asq2!iOpy29d attempt #1
4444 : WIFI : Not configured in Station Mode!!: Asq2!iOpy29d

I am using a WiFi password with special characters (#,), when I change the password to ie "test1234"it works.
:-( unfortunately I have a lot of IoT devices with coded password so I can not change password.

TD-er, 2019-01-24:
Just to be clear, you have both the `#` and the `,` in the password?

tertel, 2019-01-24:
Yes, both, comma and cross

datenheim, 2019-10-14:
Hm, something similar true for me.
My WPA key contains special characters _/\#!$ and has 62 characters in total.

Since WPA accepts nearly all printable characters for the key is there a reason to exclude some chars?
Or asked the other way: 
which (special)-characters are allowed right now for
- SSID
- WPA key

and what is the max. length supported for both?

In my case the connection attempt to the primary WIFI (Sonnentau) fails with **'(2) Auth expire'** for some reason, but the fallback WIFI (Mondtau) works as expected.

```
INIT : Booting version: mega-20191003 (ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support)
67 : INIT : Free RAM:33352
68 : INIT : Warm boot #5 Last Task: Background Task - Restart Reason: External System
71 : FS   : Mounting...
95 : FS   : Mount successful, used 75802 bytes of 957314
481 : CRC  : program checksum       ...OK
516 : CRC  : SecuritySettings CRC   ...OK
622 : INIT : Free RAM:30232
624 : INIT : I2C
624 : INIT : SPI not enabled
641 : INFO : Plugins: 46 [Normal] (ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support)
745 : WIFI : Set WiFi to STA
778 : WIFI : Connecting Sonnentau attempt #0
2409 : WD   : Uptime 0 ConnectFailures 0 FreeMem 27784 WiFiStatus 6
5658 : WIFI : Disconnected! Reason: '(2) Auth expire' Connected for 4745 ms
5774 : WIFI : Connecting Sonnentau attempt #1
10656 : WIFI : Disconnected! Reason: '(2) Auth expire' Connected for 4747 ms
10772 : WIFI : Connecting Mondtau attempt #2
14734 : WIFI : Connected! AP: Mondtau (CA:0E:14:C9:94:77) Ch: 11 Duration: 3846 ms
14735 : WIFI : DHCP IP: 192.168.179.39 (E32-meter-0) GW: 192.168.179.1 SN: 255.255.255.0   duration: 26 ms
14741 : Webserver: start
14743 : firstLoopConnectionsEstablished
32114 : WD   : Uptime 1 ConnectFailures 0 FreeMem 23736 WiFiStatus 3
```

TD-er, 2019-10-14:
Max length is 63 bytes (trailing 0)
See:
https://github.com/letscontrolit/ESPEasy/blob/79975d385bc2353f4f4297249fff1d850b962309/src/src/DataStructs/SecurityStruct.h#L16-L19

datenheim, 2019-10-14:
Ok. Reflashed with ESP Easy Flasher and NOT using post flash action to add unit name/naumber/admin-pass and wifi setting, setting only per config page works. So maybe Easy Flasher did call commands wrong.
What is the right order of commands for configuring and connecting to wifi via the serial console?
Just:
1. WifiKey "*"
2. WifiSSID "*"
3. WifiConnect
Or is something else required?

TD-er, 2019-10-14:
You should at least save the config :)
Just be aware the comma and space both are seen as a parameter separator.
So either use a single space or a comma, not both.

TD-er, 2019-10-27:
Did your password also have a `'` or `[` in it?
If so, can you test with the latest code (build yourself) or use this [test build](https://www.dropbox.com/s/xczig7tzr9mlds1/ESPEasy_mega-20191016-19-PR_2667.zip?dl=0) which has the same fixes as tomorrow's build will have.

TD-er, 2019-11-04:
Is this still an issue?

tertel, 2019-11-04:
Not for me personally, I changed the password to normal characters

TD-er, 2019-11-04:
OK, then I will close it :)

closed 2019-11-04 by TD-er

GravityRZ, 2019-12-23:
are special characters still an issue?
i know it was an issue in the R120 but not sure if the latest mega acceps this
i have .%@#;

TD-er, 2019-12-23:
I think it may be possible to store the settings.
When using a command to set the pass, make sure to wrap the parameter in quotes.  (very recent builds)

GravityRZ, 2019-12-24:
ok, will try