letscontrolit/ESPEasy #2266: [TESTING] plugin only in [DEV]

michass, 2019-01-24:
I have CSE7766 with status [TESTING], but i can't find it in test bin(4096). Dev build only.
Bst Regards.
Michał

TD-er, 2019-01-24:
It is also included in the Sonoff POW build, since it is a module mainly used in that device (as far as I know)

TD-er, 2019-01-24:
What plugins do you need?
I will move this plugin to the test builds by the way, but I have to check there will be enough free RAM left, since the 'test' builds consume quite a lot of RAM.

michass, 2019-01-24:
In my POW R2 i use also OLED SSD1306 framed and DS18b20.
Yo say test consume lot of RAM, more, than dev?

TD-er, 2019-01-24:
I will move it to the test build, but only if you make a promise to me about your setup.
Please promise me, you will make 100% sure you will not make those sensor parts available for touch when the POW is connected to mains.
The Sonoff POW is not isolated from mains power, so it is quite dangerous to use sensors which may be accessible from the outside and can be touched.
Also make sure to add some transparent plastic shield on top of the OLED display, since I don't know how well isolated the display is.

michass, 2019-01-24:
It's already done, I have Electric society diploma,  about install and maintance up to 1kV.
You have my promise

michass, 2019-02-13:
Are plugins included now?

TD-er, 2019-02-13:
Not yet merged, sorry.
It is still in the PR #2235 and since there are issues with the serial port, combined with a lot of things happening outside ESPeasy, I haven't finished & merged that one.

michass, 2019-02-13:
I can't find , how to build own ROM with my set of plugins.
Only how to build Test/dev/normal and i need to wait for yours ROMs

TD-er, 2019-02-13:
The PR is here: https://github.com/letscontrolit/ESPEasy/pull/2235
And indeed there needs to be an update/extension to the docs how to build your own version, since lots of people keep asking about that.
The last test build I made for that PR is located [here](https://www.dropbox.com/s/18bdv1ck991hu2i/ESPEasy_mega-20190202-82-PR_2235.zip?dl=0)

TD-er, 2019-02-14:
The PR has been merged this morning.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er