letscontrolit/ESPEasy #2268: The pulse counter does not retain its value after a reboot

makonrus, 2019-01-25:
The pulse counter does not retain its value after a reboot (both manual and timer).
Tested on different boards with different inputs.

Budman1758, 2019-01-26:
I don't think anything retains it's value after a reboot.  Values are not retained so as to limit the write cycles to the flash chip.  Apparently this chips fail pretty fast if writing to them constantly.

ghtester, 2019-01-26:
The RTC memory could be perhaps used to keep some data even after reboot (but not power cycle).
https://www.youtube.com/watch?v=r-hEOL007nw

ironataerial, 2019-01-26:
As I understand this RTC maintains the value while in deepsleep not after a reboot. Perhaps could survive a soft reboot but (most likely) not a power cycle. Probably not worth the resources ?

TD-er, 2019-01-26:
Indeed, it should store the last values in the RTC memory and this memory should survive a reboot as long as the chip is powered.
It will erase the RTC part if 1 bytes of data are not correct, which may be due to data corruption, or something else has changed these bytes. (change in core libs?)

blb4github, 2019-01-27:
I'm confused now, to my knowledge the pulse counter did /does never keep the total value after a reboot. As mentioned in the device screen; "Total value is not Persistent!" I've never seen this value retained, the total counter is always reset to 0 after a reboot. Currently running ESP_Easy_mega-20190110_test_core_250_beta_ESP8266_4096.bin on a wemos D1 mini pro. Should it be retained (due to code to use the RTC memory)?

TD-er, 2019-01-27:
I must have a look at the code, but as far as I know there is a part in the code which should store the plugin values to the RTC memory.
But like I said, that's only to survive a hot reboot, not a power down.

dorinvlas, 2020-03-09:
@TD-er was this one solved? How we are making sure that our counters do retain data after hot reboot or power outages? 

Thank you so much!

TD-er, 2020-03-10:
The output values of plugins will be restored after a (hot) boot.
So that should work.
If you have an example of a plugin not behaving well, please let me know.
Otherwise this issue can be closed.

Tsjippy, 2020-03-10:
It is not working here.
I am on mega-20200310.
I have a "Generic - Pulse counter" device, the values are not kept during reboot, even not during a sleep cycle.

dorinvlas, 2020-03-15:
I do confirm that using an self build from 13th of March source code, running espeasy on esp32dev it is not keeping data on reboot.

TD-er, 2020-03-16:
> I do confirm that using an self build from 13th of March source code, running espeasy on esp32dev it is not keeping data on reboot.

ESP32 is not using RTC memory at all (yet), so the ESP32 will not keep any values at reboot.

TD-er, 2020-03-19:
Could you try [ESPEasy_mega-20200310-74-PR_2952.zip](https://www.dropbox.com/s/w34c2fqv0liifyb/ESPEasy_mega-20200310-74-PR_2952.zip?dl=0) to test if the pulse counter may retain its values after a reboot?

blb4github, 2020-03-19:
Hi TD-er, Ik can confirm with this version pulse counter does retain it's values now after a reboot! Thanks!

Now it becomes also more necessary to be able to reset the values with a command which is not working if you have more than one pulse counter on a module.

dorinvlas, 2020-03-19:
> Hi TD-er, Ik can confirm with this version pulse counter does retain it's values now after a reboot! Thanks!
> 
> Now it becomes also more necessary to be able to reset the values with a command which is not working if you have more than one pulse counter on a module. see #2710

Did you test it on esp32?

TD-er, 2020-03-19:
> Did you test it on esp32?

I can guarantee you that will not (yet) work, as our ESP32 build  has not yet support for RTC memory.

Tsjippy, 2020-03-20:
> 
> 
> Could you try [ESPEasy_mega-20200310-74-PR_2952.zip](https://www.dropbox.com/s/w34c2fqv0liifyb/ESPEasy_mega-20200310-74-PR_2952.zip?dl=0) to test if the pulse counter may retain its values after a reboot?

I did test it on a Wemos Pro mini, and it works. Thanks!

blb4github, 2020-03-20:
> I can guarantee you that will not (yet) work, as our ESP32 build has not yet support for RTC memory.

I did test on a Wemos Pro mini as well.

commit 7d4dd235 (2020-03-21): Pulse Counter must retain its values after reboot (#2268)

Fixes #2268
closed 2020-03-21 by TD-er