letscontrolit/ESPEasy #2271: nodemcu-tls

ilsicca, 2019-01-27:
Hi!
**Problem**: I need to connect to an MQTT broker (port 8883) with TLS.
I would like to know if there is any version of ESPEasy for Nodemcu ESP-12 that allows me to configure TLS towards MQTT by adding my certificate?

TD-er, 2019-01-27:
At this moment there is not a version of ESPeasy that does handle TLS.
The memory of the ESP8266 is just to small to handle.
But I think we can add it to an ESP32 build.
Maybe the ESP32 can then also act as a hub between ESPeasy nodes and the MQTT broker, but for now there is no such possibility.

Maybe Mosquito or some other package can act as a conversion between "insecure" and TLS transport?

ilsicca, 2019-01-27:
I will wait for the build with TLS or I will look for another solution, thanks!

TD-er, 2019-01-27:
Maybe you could also try something like this [stunnel.org](https://www.stunnel.org/)
It kinda defeats the purpose of a secure layer, but maybe worth a try.

denisfrench, 2019-12-14:
Can we re-open this issue as a feature request until resolved please?

Somehow now both Tasmota and ESPurna manage to include MQTT with TLS on the ESP8266. I think I saw somewhere the arduino core library recently changed the TLS implementation to be less resource intensive.

See also #617, where MQTT TLS support is a core issue.

deki, 2020-05-02:
Tasmota implementation notes https://tasmota.github.io/docs/TLS/#implementation-notes give some insights on BearSSL.

Is it maybe an option to implement it but not enable it in the default version like ESPurna does it?

TD-er, 2020-05-02:
@deki I think that's an option now.
The BearSSL is a lot more easy on the resources, so maybe we can try it now again.

deki, 2020-05-13:
If you are looking for a beta tester, let me know...

closed 2024-10-02 by TD-er