letscontrolit/ESPEasy #2272: [feature request] rules visual UI

ppisljar, 2019-01-28:
it would be really nice if we would have an UI, similar to node-red, for setting the rules.

it should:
- have a list of available triggers
- have a list of available actions
- list of available functions (IF, is there anything else ? )

by dragging and dropping the above nodes you could visually construct your rules.

has anyone started working on something similar already ? else i might be able to do some PoC over the next weekend.

is there any space left on the device with current MEGA release ? we would need to store html&javascript for the UI, even without using any external libraries we'll be hitting 100kb probably

Grovkillen, 2019-01-28:
We're not going to include that inside the actual nodes themselves but the idea is to have a portal for more advanced editing. We're currently developing that as we speak.

LeeNX, 2019-01-28:
I have been thinking of the idea of using a portal or something to do all the heavy lifting for configuring type stuff. Awesome to know that somebody has already thought of this and is working on it. Thanks Guys!!!

ppisljar, 2019-02-03:
i worked on this a bit. locally the editor is functioning and produces valid filters.
I tried to then load it to espeasy (using regular upload, as i can upload pages there)
the upload completed, i can see the file listed under flash ... but trying to load the file doesn't work ... server (espeasy) doesn't reply.

i am wondering what could be wrong ? Does the parser crash, due to overcomplicated html file ? Or is the file too big ? (doesn't seem the problem, as config.dat is way bigger than this)

currently the whole editor, + basic nodes + connectivity with espeasy comes down to 14kb minified.

simple configuration:
[image]

simple filter output:
```
  on Rules#Timer=1 do
    Delay 500
    If [TEMP#Output]=1
      gpio,12,1
      timerSet,1,60
    Else
      timerSet,1,5
      gpio,12,0
    Endif
  Endon
```

[editor.zip](https://github.com/letscontrolit/ESPEasy/files/2825828/editor.zip)

TD-er, 2019-02-03:
ESPeasy is running the webserver itself.
So for every URL it can handle, there must be a description in the source code to tell the webserver what to do with the file.
So I guess there is no matching entry for the file you store to the SPIFFS, or it isn't serving the file with the proper flags.

ppisljar, 2019-02-03:
and yes, encoding is ANSI ASCII

ppisljar, 2019-02-03:
uploading as `.htm` seems to do the trick ...

ppisljar, 2019-02-03:
installation instructions:
- unzip
- go to espeasy -> tools -> load
- select editor.htm file that was unziped
- click upload
- after upload is complete go to http://your-esp-ip/editor.htm

warning:  clicking SAVE will override your rules1.txt

ppisljar, 2019-02-03:
some updates: 
[editor.zip](https://github.com/letscontrolit/ESPEasy/files/2825897/editor.zip)

Grovkillen, 2019-12-29:
https://github.com/letscontrolit/GUIEasy

I will go with the "old" look and feel of the Rules editor but I have added syntax highlighting. Will add more of the initial idea (first post).

After we have the new GUI deployed I will start working on a more elevated rules editor.