letscontrolit/ESPEasy #2273: ThingSpeak requires MustCheckReply = True

jasontwinters, 2019-01-30:
For requests to be processed by ThingSpeak.com, MustCheckReply will need to be set to True. 

I am a developer working for ThingSpeak.  A permanent change to our system architecture has the unintended consequence of ignoring requests from clients which close the TCP connection immediately following the HTTP GET or POST.  In order to have requests properly processed by ThingSpeak, the client needs to wait for a response. We have confirmed this is our testing.  The change will go live the first week of Feb-2019.

TD-er, 2019-01-31:
So that change is going live tomorrow?

jasontwinters, 2019-01-31:
Feb 4, 2019 is the target date for the change on ThingSpeak.

We did a test run for a few hours a month ago and we saw bunch of letscontrolit requests failing.  I'll be honest, I'm not familiar enough with ESPEasy code to change it myself and request a merge.  But from spending an hour looking at the code I can see that MustCheckReply needs to be set to true in order for it work going forward.  I'm not sure if you do that at the _C004.ino level or something higher.

My hope was to alert you of the upcoming problem before it hit.

TD-er, 2019-02-11:
See also replies here:
https://github.com/letscontrolit/ESPEasy/issues/2289#issuecomment-462188764

Grovkillen, 2019-02-14:
We need to have the settings default to acknowledge reply and then we can close this right?

TD-er, 2019-02-14:
I guess so

thomastech, 2019-08-19:
I believe this issue (MustCheckReply=True) is affecting sendToHTTP. Please long discussion at https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=6954.

sendToHTTP uses MustCheckReply=false. This is hard coded in the function.

- Thomas

kuzma00, 2019-12-23:
Add this information to the wiki page, please. It was difficult to find it.

TD-er, 2019-12-23:
> Add this information to the wiki page, please. It was difficult to find it.

You mean for the sendToHttp command?
That check reply option is less than a week old.

For the controllers, it is already present in the [Read The Docs](https://espeasy.readthedocs.io/en/latest/Controller/_Controller.html#controller-parameters) but I do admit it may need its own chapter.

kuzma00, 2019-12-24:
I mean settings for thingspeak. Check Reply: check Acknowledgment

TD-er, 2019-12-25:
Do you have a link of the docs where it should be added, or at least where things are unclear?

kuzma00, 2019-12-25:
https://www.letscontrolit.com/wiki/index.php/EasyProtocols#ThingSpeak

TD-er, 2019-12-25:
Ah the wiki... well we plan on move the documentation to the Read The Docs, so the wiki has not been maintained for a long time now.