letscontrolit/ESPEasy #2274: Plugins with ID>255 don't appear in the device pick list

BNNorman, 2019-01-31:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

BNNorman, 2019-01-31:
I can't update the initial post so here goes.

**Summarize of the problem/feature request**
When I add a new plugin with an ID>255 (I used 300 for testing to avoid clashes) and add it in define_plugin_sets.h. The platformio compile succeeds with a warning Device[++DeviceCount].Number being truncated (presumably to a uint8_t) , which can be missed if not sat watching the compile. After flashing the resulting bin and going to the devices dropdown list the device is not in the list.

**Expected Behaviour**
The compile should fail since Device[++DeviceCount].Number will probably clash with a plugin seen before. But I doubt the compiler would pick that up.

**Actual behavior**
The compile succeeds but the device does not appear in the drop down list.

**Steps to reproduce**
Just modify any plugin and set the ID etc to 300 - I'm guessing 256 is the upper limit (not seen that doucmented)

**System configuration**
Desktop? Windows 10 64 bit

**Hardware:**
nodeMCU V1.0 ESP8266

**ESP Easy version:**
Self build based on Mega 20190121

**ESP Easy settings/screenshots:**
Not needed - can't capture the compiler output (or I don't know how - noob).

**Rules or log data**
N/A

TD-er, 2019-01-31:
Hmm that would mean a breaking change in the settings.
Not sure yet how we should overcome this limit.
Reminds me of the Y2k bug which already makes me feel old ;)

BNNorman, 2019-01-31:
Already playground plugins are over the 199 and, within define_plugin_sets.h there is a reference to P213. The playground limit of 199 is the only one I am aware of and that has already been exceeded. It's an interesting problem.

TD-er, 2019-01-31:
The id of the plugin is an unsigned 8 bit value, so it will need a change in the settings struct to make it work with bigger numbers and I'm not sure at how many places this assumption is being made.
So to fix it we may need to change a lot of files and existing settings will no longer be compatible.

BNNorman, 2019-01-31:
I must admit I haven't looked at the code (Just got my plugin working) Can the plugin IDs be aliased in some way? Devices[] contains sequential numbers perhaps the index could be an alias? It is unlikely a build will contain 256 plugins but there may be a lot more come out of the woodwork in future. I did a quick search of the src folder but I couldn't find where Device[..] is defined or used but I imagine it's in a lot of places.

tonhuisman, 2022-05-07:
This seems outdated, so can be closed

closed 2022-05-07 by TD-er