letscontrolit/ESPEasy #2277: unable to flash any .bin file

trudeau102000, 2019-02-02:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

trudeau102000, 2019-02-02:
when i select a file to flash and click flash all i get is an instant message saying flash complete but nothing flashes to module

Grovkillen, 2019-02-02:
What do this issue mean?

Grovkillen, 2019-02-02:
Are you using the "ESP.Easy.Flasher"?

trudeau102000, 2019-02-02:
i am using the flash8266 application in ESPEasy

Grovkillen, 2019-02-02:
Would you mind testing the official Flasher?

trudeau102000, 2019-02-02:
[image]

Grovkillen, 2019-02-02:
Is the serial port correctly installed? The icon suggest you're having some issue with it.

TD-er, 2019-02-02:
Just some additional tips:
- Use a powered USB hub
- Use proper quality USB cables
- Sometimes a full power-down/reconnect of an USB hub may solve a lot of strange issues.