letscontrolit/ESPEasy #2281: PLUGIN_DESCR overwrites DEFAULT_NAME

clumsy-stefan, 2019-02-03:
In
https://github.com/letscontrolit/ESPEasy/blob/d40c449f8d825c23fb1f93253bf17fb80370eba9/src/Misc.ino#L1192-L1194
The defined DEFAULT_NAME gets overwritten by "PLUGIN_DESCR" which also seems to affect the default AP-SSID in a fresh config. As PLUGIN_DESCR likely contains special characters, spaces, etc. it's not ideal as SSID. Just IMHO...
The default SSID should be individually defined (eg. DEFAULT_NAME) and the PLUGIN_DESCR only shown on the sysinfo page!

clumsy-stefan, 2019-02-03:
Not a big issue really (if one knows about it), but I'd suggest to either make a new define fro the default SSID or just remove these lines and keep DEFAULT_NAME from
https://github.com/letscontrolit/ESPEasy/blob/d40c449f8d825c23fb1f93253bf17fb80370eba9/src/Misc.ino#L1127

tonhuisman, 2022-04-18:
This seems to be resolved some time ago, so can probably be closed.

closed 2022-04-18 by TD-er