letscontrolit/ESPEasy #2282: Calling deepsleep within a rule does not disconnect from MQTT

nuccleon, 2019-02-04:
Hi all,

I've an Issue with forcing deepsleep from within a Rule with **Release mega-20190202**

Activating deepsleep  by defining 
> Sleep awake time 20 [sec]
> Sleep time 600
in the confic section works fine. That means:
- Disconnectiong from MQTT
- Disconnect WiFi
- Go into sleep

Activate Deepsleep within a rule works different
```
On Rules#Timer=1 do  // going to sleep for 600 seconds
deepsleep,600
endon
```
- Go into sleep
- Broker publish LWT

No mqtt disconnection, no wifi disconnection. That forces the broker to publish LWT :-1: 

Is there a way to work around this issue? 
Something like a command for disconnection MQTT?

TD-er, 2019-02-04:
Sounds like a bug to me.
No matter who/what is calling a command, it should show the same behavior.

TD-er, 2022-04-01:
This is no longer an issue, as a long time ago the MQTT disconnect was added when calling deep sleep.

closed 2022-04-01 by TD-er