letscontrolit/ESPEasy #2283: P078 EASTRON SDM120C receiving NAN,

D-t-M, 2019-02-04:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2019-02-04:
I wonder why we still try to put more hints in the issue template to get the minimal amount of information for a new issue.

I need:
- version that was working and version that is giving this issue.
- Some indication on how often this NaN value occurs.
- Some indication on how it is connected (length of cable, type of RS485 adapter used, what pins are connected, pullup-/pulldown configuration of those pins)

That's the absolute minimum to be able to do something with this issue.

D-t-M, 2019-02-04:
Sorry for the first empty post. I hit the wrong buttons and the still empty post was online.
I tried to fill out more information using the edit function!

D-t-M, 2019-02-04:
Hello gain,

OK, now all the information should be online.
Sorry again for the inconveniences!

What I additionally tried was using the hardware serial (RX/TX) lines, which causes only one single reading from the first value and then stopped at all. I even did noch see any flickering of the TX LED on my RS485 interface. But this is maybe a different topic.

Best regards,
D-t-M

TD-er, 2019-02-04:
I am adding CRC pass/fail notes to other plugins using serial communications too, to get an idea on how often those fail.
For the GPS plugin it is near impossible to use software serial because of the amount of failed messages.
Those failure rates are between 50 and 90%, but those messages are longer and the GPS is quite chatty.
When using HW serial the number of read lines with CRC failures is near zero.
For example one of my test nodes has handled 5 million lines with 14 CRC errors using HW serial.

Can you use HW serial using GPIO 13 and 15?  (That's "Serial0 swapped")
Then you have to disable the serial log to make it work.

D-t-M, 2019-02-04:
I tried to use the hardware serial with the swapped pins, too, but when rebooting the node it got stuck, because there is a pull-up on one of the lines in my RS485 interface.

But maybe I can try connecting the RS485 interface to the GPIO 13 and 15 after booting for testing…. give me a few minutes, please.

TD-er, 2019-02-04:
Hmm, that's right GPIO15 is a bit tricky at boot.

TD-er, 2019-02-04:
If that may be working, then you could try to add some circuitry to only connect the RX of the modbus when boot has finished.
I'm afraid that will cost you another GPIO pin, but it may be worth it?

D-t-M, 2019-02-04:
Ok, I hooked the RS485 interface to GPIO13 and GPIO15, reconfigured the device but I don´t get any transmission at all (no RX and TX LED light up, all values NAN).

I disabled the "serial output" in the Advanced options already to not interfere with the Modbus communication.

Somehow it looks like the hardware serial is not working at all in the Plugin. It does for debugging with an USB cable and a terminal running in my laptop.

TD-er, 2019-02-04:
Are you sure TX and RX are the correct way?
GPIO15 should be connected to the RX of the sensor/RS485 adapter.
You may also want to have a look at the pull-up configuration.
See also our documentation [GPIO - best pins to use on esp8266](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266)
GPIO15 is a bit strange, since it may be actively pulled down on your board.

You can also try the normal Serial0 pins (GPIO 1 & 3) which do not have this strange config.
But then the sensor may see some boot logs and there is no guarantee what will be set/changed on the connected Modbus slave or sensor.

D-t-M, 2019-02-04:
Yes, RX of the RS485 is connected to GPIO15, and TX of the RS485 interface is connected to GPIO13. Both lines are showing no activity at the LED on the interface.

When using the RX/TX lines - even if there is some traffic during booting - I at least see the TX LED blinking once. Compared to the SoftSerial config I see this TX LED blinking 4 times followed each by one blink of the RX LED.

So I think the HW Serial 0 seems to send only one request to the Modbus. I even can see the very first value presented followed by NANs.

TD-er, 2019-02-04:
When the transmission to the sensor is not occurring, then there will also be no data sent the other way around.
It is a master/slave protocol, so you have to actively request the data.
So that makes sense.

I will have a look at the code for GPIO15 in the ESPeasySerial library.
Some user also reported the Nextion plugin was giving issues with these changes, so it is possible there actually is a bug in there.

D-t-M, 2019-02-04:
I flashed ESP_Easy_mega-20181231_test_core_250_beta_ESP8266_4096_VCC.bin
The config stayed unchanged and the RS485 interface is connected to RX/TX (GPIO 1 and 3).
All 4 values are showing up. I´ll keep it running like this and see, if there are any NAN values.
Seems like there is something different between these two builds.

TD-er, 2019-02-04:
There is a lot of changes between those builds, since I introduced the SW/HW serial abstraction then
But now I come to think of it, I only tested with sensors that will read on the HW serial

D-t-M, 2019-02-04:
OK, let me know when I can help with some tests or something like this.
Pooringly my programming skills are very basic and I even did not manage to compile my own build yet. So I guess I can´t help on that level :-( 

Thank you very much for your work and your support! 
I discovered ESPeasy 4 weeks ago and I got a fan immediately, running 7 nodes already with OLED displays, temperature and humidity monitoring and another energymeter using S0 pulsecounter already :-)

TD-er, 2019-02-04:
I think I will split those plugins really needing the HW serial into another PR and then revert those changes back to the 20181231 build.
That will make a stable ground to work on and try to fix what's apparently wrong with the TX line.

I just opened a new beer, so maybe I can see the bug later this night, but if not then I will split those and revert the SDMxx and Nextion plugin to 20181231

D-t-M, 2019-02-04:
Enjoy your beer and good luck with bug hunting ;-)

TD-er, 2019-02-04:
Probably found the problem.
Let's hope I can make some fix and then I will make a new test build first.

D-t-M, 2019-02-05:
Hello,
I uploaded the test build and get values from the SDM120 using the HW serial 0 (GPIO 1 and 3). No NAN values since 10 minutes now.
Using the swapped pins (GPIO 13 and 15) i see no activity in the TX LED at all.
I´ll have a closer look at this this evening after I'm back from office.
Thank you very much for the test build!

TD-er, 2019-02-05:
So at least for GPIO 1&3 (Serial0) it is now working and Serial0_swapped not yet.
For GPIO15 it may be an issue with pull-down of the pin on the board (it has to be low during boot)

While connected to GPIO 1&3, do you have any issues during boot?

fluppie, 2019-02-05:
I use a SDM220-MODBUS on D5 and D6 with the 17 12 2018 build. I'll try to update to the newest and check if it stops working.
[image]

fluppie, 2019-02-05:
That went wrong quickly :)
[image]
Looks like I'll revert to 18 12 2018. I also have issues with the newer builds and senseair s8 sensors. I use 18/12/2018 for them as well.

fluppie, 2019-02-05:
With your special test build:
[image]

TD-er, 2019-02-05:
I am looking into the SenseAir plugin as we speak.
One of the issues with it was that it didn't check the CRC of incoming messages.

About the Eastron plugin. The main change with the test build I made last night was to be able to use HW serial again (was broken since 20190101).
SW serial is still a bit unreliable so it is not really strange to see a NaN every now and then.
I am looking into the other plugins using serial port, so I will also have a look at this one.

fluppie, 2019-02-05:
I agree, but with these low baud rates of 2400/4800/9600 I think it should work just fine?

TD-er, 2019-02-05:
My GPS plugin is also working on 9600 baud, but on SW serial the failure rate is 50 - 90%
Those messages are a lot longer (upto 80 bytes) so there is a higher chance to miss a bit.
HW serial is working a lot better (~10 fails in 5 million lines)

It also depends on other activity of the node.

I will add an indicator in the Eastron plugin to see how often a CRC error is detected in the Modbus communication.

TD-er, 2019-02-05:
I just added the CRC pass/fail stats to the plugin page (not tested)
Also the Modbus call to read a value may retry up-to 3 times to get a value if there's something wrong.

See this commit: https://github.com/letscontrolit/ESPEasy/pull/2235/commits/3b1dd8f21f05fd047c35d1bb2753438df64edca4
I am making a test build as we speak for this. It will be ready in about 30 minutes.

TD-er, 2019-02-05:
[New test build](https://www.dropbox.com/s/gpjz8286m5cycmt/ESPEasy_mega-20190202-65-PR_2235.zip?dl=0)
It is including the linked commit of my previous post.

fluppie, 2019-02-05:
OK, it's running for 1minute now: Checksum (pass/fail): | 40/7

TD-er, 2019-02-05:
I guess the 3 retries now take care of hiding the NaN values?
Or did you see them still?

fluppie, 2019-02-05:
Looks like I don't see them.
At the moment:  Checksum (pass/fail): | 936/158

TD-er, 2019-02-05:
I started to build te same setup on my bench, so tomorrow I can also test myself and will also look at the signals with the scope to see what's happening. (e.g. maybe a pull-up config needed or internal pull-down disabled)

fluppie, 2019-02-06:
I see no NaN's in my log :).
Checksum (pass/fail): | 6080/1021

TD-er, 2019-02-11:
At least the NaN part of this issue will be fixed when I merge my open PR #2235 
The HW serial part is still an issue, but that may also be related to the used 485 converter board I used. (pull-up/pull-down resistor issue)
So if you don't mind, I close this one for now and the HW serial will also be dealt with in other issues.

closed 2019-02-11 by TD-er

D-t-M, 2019-02-12:
Hello,
that´s ok for me, closing this issue here.
2 Questions:
- Is the counting of the CRC errors also included in the new release?
- Can you give me an issue number about the HW serial issues, that I can watch? 
My targeted solution is using the HW serial, which runs good with the 20181231 release. Pooringly this release resets the device 10 times a day. That´s why I am looking forward to a newer release with HW serial running :-)

Thank you for your help!

Best regards,
D-t-M

D-t-M, 2019-02-12:
found the reference after removing the tomatoes from my eyes :-)

Both questions are answered! Thank you!

D-t-M, 2019-02-19:
Hello,
to give an feedback on the release 20190116 I tested:
This release is running with HW serial on D9/D10 stable now for 21 hours. I had no reboots (as I had with release 20181231) and there were only 6 CRC fails. The read-counter had an overflow already, now it is at 800 times, but it was at 58000 today already. ==> perfect!

So this release seems to have fixed the hw serial configuration :-)

[image]
Build:⋄
20103  - Mega
Libraries:⋄
ESP82xx Core 2.6.0-dev, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support
GIT version:⋄
mega-20190216
Plugins:⋄
76 [Normal] [Testing]
Build time:⋄
Feb 16 2019 03:24:39
Binary filename:⋄
ESP_Easy_mega-20190216_test_core_260_alpha_ESP8266_4096_VCC.bin

fluppie, 2019-02-20:
I'm running mega-20190216
Checksum (pass/fail): | 6336/0

Only a uptime of4 hours, but it has been running for 3 days.
Boot: | Manual reboot (4)
Reset Reason: | Hardware Watchdog