letscontrolit/ESPEasy #2286: 1MB build with P035 IR plugins

DoctorOctagonapus, 2019-02-05:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it. **N/A**
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective. **N/A**
- [x] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

As discussed in #1790, current build with P035 IR plugins (ESP_Easy_mega-xxxxxxxx_normal_IR_ESP8266_4096) is only compiled for 4MB devices. Requesting a 1MB build for ESP-01 or similar if at all possible please.

For my particular use case the DHT plugin would be useful to have if there is room for anything in the build, but if that can't be done it can't be done.

### Expected behavior

N/A

### Actual behavior

N/A

### Steps to reproduce

N/A

### System configuration

Hardware: ESP-01

ESP Easy version: All versions since 2018-08-22

ESP Easy settings/screenshots: N/A

TD-er, 2019-02-14:
It has been added and will be included in the next nightly build.
Not sure how long it will last without any issue, since it is already taking a lot of the available space.
```
DATA:    [======    ]  62.3% (used 51024 bytes from 81920 bytes)
PROGRAM: [==========]  97.7% (used 872236 bytes from 892912 bytes)
```
If it doesn't fit in `PROGRAM` space at some point, it will not be included in the nightly build ZIP file.
So if it is missing at some point, we may need to strip some plugins from this build to make it fit again.

DoctorOctagonapus, 2019-02-15:
Appreciate it. I've just downloaded the latest nightly build, am I right in thinking it's the normal_ESP8266_1024_VCC or normal_ESP8266_1024 I'm after? The only one marked IR is the 4MB build.

TD-er, 2019-02-15:
Hmm I have to look into the before_deploy script to see if I added it there too.
Maybe I only added it to the platformio.ini file and then it will not be included indeed :(

DoctorOctagonapus, 2019-02-20:
Hey, the new build flashes and boots fine, and the option to add an IR device is there. Haven't had a chance to get my head around sending commands to it yet, but for now at least it seems to be working. Thanks for that.

closed 2019-02-20 by TD-er