letscontrolit/ESPEasy #2287: [feature request] add gpio and port to /json output

ppisljar, 2019-02-06:
We can access list of devices in json format over http://ip-to-esp/json 
However, in comparison to the device list on the webpage its missing the GPIO and port fields. 

It would be great if those could be added.

TD-er, 2019-02-06:
You mean this section of the JSON:
```json
{  
   TaskValues:[  
      {  
         ValueNumber:1,
         Name:"Switch",
         NrDecimals:0,
         Value:1
      }
   ],
   DataAcquisition:[  
      {  
         Controller:1,
         IDX:0,
         Enabled:"false"
      },
      {  
         Controller:2,
         IDX:0,
         Enabled:"false"
      },
      {  
         Controller:3,
         IDX:0,
         Enabled:"false"
      }
   ],
   TaskInterval:0,
   Type:"Switch input - Switch",
   TaskName:"btn",
   TaskEnabled:"true",
   TaskNumber:2
}
```

ppisljar, 2019-02-06:
yes, in this section it would be nice to see the GPIO, when relevant (as we do on the devices page)

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er