letscontrolit/ESPEasy #2288: Rule Publish sends to first controller even though it is disabled

BNNorman, 2019-02-06:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x ] This issue describes what is happening.
- [x ] This issue describes what components are affected (e.g. name of plugin/controller)
- [x ] This issue describes how to reproduce it.
- [x ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x ] given a list of active plugins or controllers when applicable.
- [ x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [x ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Whilst developing using a self build of Mega I used a local broker and set controller #1 to send MQTT messages to it. These were captured into my local database for checking. When I went live I added a second (remote) controller enabled it and disabled the first (So I could re-enable it if need be). Booted the MCU and everything was fine for a few hours. Messages went to the second controller as expected. Later in the day I discovered messages were not going to the active controller but found they were now going to the inactive (local test) controller. 

### Expected behavior

Publish should go to the active controller only.

### Actual behavior

Initially Publish sent data to the active controller (2nd in list) but switched over to using the inactive controller (first in the list).

### Steps to reproduce

1. Add two controllers and disable the first.
2. Select a device, make it active but do not select a controller.
3. Create a periodic rule to publish data (I need to send a JSON string)

The system works as expected after a reboot.

### System configuration

Hardware: nodeMCU V1.0, sds011 DUST SENSOR

Sysinfo page
Unit: | 4
-- | --
Local Time: | 2019-02-06 19:42:39
Uptime: | 0 days 0 hours 9 minutes
Load: | 6.10% (LC=5997)
Free Mem: | 12592 (11832 - sendContentBlocking)
Free Stack: | 3616 (1756 - ExecuteCommand)
Boot: | Manual reboot (2)
Reset Reason: | Software/System restart
Network ❔
Wifi: | 802.11N (RSSI -74 dB)
IP config: | Static
IP / subnet: | 192.168.1.83 / 255.255.255.0
GW: | 192.168.1.1
Client IP: | 192.168.1.10
DNS: | 192.168.1.1 / 0.0.0.0
Allowed IP Range: | 192.168.1.0 - 192.168.1.255
STA MAC: | 80:7D:3A:74:F1:2C
AP MAC: | 82:7D:3A:74:F1:2C
SSID: | KC Lightstream 10 (58:8B:F3:3E:CB:3B)
Channel: | 9
Connected: | 8 m 44 s
Last Disconnect Reason: | (1) Unspecified
Number reconnects: | 0
Firmware
Build:⋄ | 20103  - Mega
Libraries:⋄ | ESP82xx Core 2_4_1, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
GIT version:⋄ |  
Plugins:⋄ | 47 [Normal]
Build time:⋄ | Jan 30 2019 23:27:55
Binary filename:⋄ | Self built!
System Status
Syslog Log Level: | Error
Serial Log Level: | Error
Web Log Level: | None
ESP board
ESP Chip ID: | 7663916 (0x74F12C)
ESP Chip Freq: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP12E
Storage
Flash Chip ID: | Vendor: 0x20 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE speed: | 40 MHz
Flash IDE mode: | DIO
Flash Writes: | 2 daily / 21 boot
Sketch Size: | 785 kB (2284 kB free)
SPIFFS Size: | 934 kB (860 kB free)

ESP Easy version: self build of Mega - Official builds don't have the ZHP01 sensor - I had to write a driver for it but this occurs with the SDS011 device.

[image]
The screenshot above shows the route taken by the data. I have two python subscriber scripts running on a Pi. The one listening to my local broker appends 'local_broker' to the topic so I know how the message gets to my database (for testing this problem only). From the remote broker I append CH to the topic. The screen shot shows the payload with dev=brian04 was recieved by my local broker - it should have gone to the remote in which case my subscribe script would have added CH.

ESP Easy settings/screenshots: 
This shows the SDS is not configured to Send to either controller (The rules do it)
[image]

### Rules or log data

on MQTT#Connected do
timerSet,1,5
endon

on Rules#Timer=1 do
Publish airquality/data,{"dev":"brian04","PM25":[SDS011#PM2.5],"PM10":[SDS011#PM10]}
timerSet,1,360
endon

```

```

TD-er, 2019-02-11:
I finally have time to look at these issues....
You're referring to the values via their name/value notation.
So it is fetching what is also displayed in the devices page.

This means this issue is actually a design discussion on how to handle disabled plugins.
Should they keep the last known value, or should it be "NaN" ?

The route of sending through a controller is already taken care of, since that will only be called from the plugin if it returns a value when calling the "READ" function.
But here you're referring to a sensor/value couple, so instead of "push" you're "pull"-ing the value.
Thus leaves the question, what should a disabled plugin have as last known value?
And the related question; What value should a plugin return when no data was read or an error occurred when reading?

BNNorman, 2019-02-11:
No, the issue isn't the values - they were fine but being sent via a disabled controller. Oddly, the problem occurred at about 2am - a bit difficult to catch when asleep LOL.

The fact is that after a reboot the data was correctly sent to the enabled controller. 

When I was developing/checking the rules I had the first controller(0) enabled. The MCU sent the data to my debug broker. I then added a new controller (1) for my live broker. Disabled my test controller, enabled the live controller and every thing was fine till the middle of the night when ESPeasy reverted to sending to my test controller (controller 0).

Meanwhile, I have changed the first controller to point to my live broker so ESPeasy no longer has a choice as I cannot afford for the live data to go missing again. (Air Quality monitoring in my City).

I had a go at following your code but the routine to select the first enabled controller looks sound to me.

I guess I could put mosquitto on another Pi and try to repeat the problem.

TD-er, 2019-02-11:
There is one thing you may need to know when searching for things like these.
ESPeasy has a fall-back option when it is failing to boot a number of times.
For example when a plugin or controller is running on corrupt settings or there is a bug in there.
After 10 unsuccessful boots, it will start disabling plugins and controllers, one at a time, until it can complete a boot.
So it might be your setup is crashing repeatedly for some reason and suddenly is able to complete when the first enabled controller is disabled.

That may explain the behavior you're seeing.

BNNorman, 2019-02-11:
I don't recollect seeing any published messages to that effect. Normally I would expect to see a 'connected' message after a reboot. As I said - I'll try to setup a test with a second Pi to see if I can repeat it.

BNNorman, 2019-02-18:
I've been unable to repeat this so I'm closing the issue. One of life's glitches I guess.

closed 2019-02-18 by BNNorman