letscontrolit/ESPEasy #2289: posting to thingspeak Not posting all IDX

Ryanmaturi, 2019-02-07:
Posting from latest release of esp easy to thinkspeak values of SI7021 under IDX 1  and 2 works great waking from deep sleep every 15 minutes.  when i add in an analog read using voltage dividers under IDX 3 it only posts it one time to the channel when i reboot the device. why is it not posting to all 3 IDX ? 
i have the sensor delay set to 60 seconds on all 3 of the devices.

Grovkillen, 2019-02-07:
What is your controller settings?

TD-er, 2019-02-07:
Like what Jimmy said, please show the controller settings and also what kind of subscription you have for Thingspeak.
Please also have a look at the [Controller settings documentation](https://espeasy.readthedocs.io/en/latest/Controller/_Controller.html?highlight=thingspeak)

Also (maybe not related at all), please have a look at this issue: https://github.com/letscontrolit/ESPEasy/issues/2273

TD-er, 2019-02-07:
It could be the buffers are not flushed well before going to deep sleep?
The minimal send interval is compared to the timestamp of the last sent message by that controller.
If the device is already going to deepsleep before that queue is emptied you may get some strange results.

Can you test to change the order of the items being sent, just to see if it is always the last one that's not working well?

Ryanmaturi, 2019-02-07:
Where is this done?  I think you may be onto something as I have played around with it and had only the idx #3 values show up and not the 1 and 2 at a point

Ryanmaturi, 2019-02-08:
I changed the order.  Still not receiving analog value on thing speak.  Should I try a different version of espeasy

Grovkillen, 2019-02-08:
@TD-er, is that request reply thing kicking in you think? We're dropping the connection without a handshake, right?

TD-er, 2019-02-08:
It might be, that's why I referred to issue #2273

Ryanmaturi, 2019-02-10:
How do I fix this ? Does it require compiling from arduino, I can do that but am lazy to get all the deps for it.  I usually flash the .bin file. Could it be added as a check box on the controllers screen ?

TD-er, 2019-02-10:
Just relating to issue #2273 
If I understood that issue correctly, you should have to enable the wait for an acknowledgement in the controller settings:

[image]

It is already present in the controller settings.

Ryanmaturi, 2019-02-11:
Problem solved. Posting all index values including battery voltage while deepsleeping.  Thanks.

closed 2019-02-11 by Ryanmaturi

TD-er, 2019-02-11:
So you had to enable this "Check Acknowledgement" option to make it work?
Then we must include it in the documentation and maybe force it to be default for ThingSpeak.

Ryanmaturi, 2019-02-11:
Correct, at least while it was In deep sleep. Also maybe consider setting the defaults for the minimum send interval and ip address. Thanks for all your help.