letscontrolit/ESPEasy #2290: On System#Boot  %systime% not working?

mlfactory, 2019-02-07:
Hello,

i have a Problem with  %systime% on Rules @ Boot.

Here is my Rule:
```
`On System#Boot do
if %systime% > 08:00:00 and %systime% < 20:00:00
   gpio,5,1
endif

endon
```

This rule should start a relay.

Problem is , it is not working.

`if %systime% > 08:00:00`

Alone also work NOT now is 14:18

`if %systime% < 08:00:00`

WORKED....

So i think on boot the Time is 00:00:00.

Is that true?

Cheers

`

Grovkillen, 2019-02-07:
Are you sure you can use those logical tests on systime? I mean, they are not floats or integers that can be tested. They are strings so only = would work.

Grovkillen, 2019-02-07:
`%syshour%%sysmin%syssec% < 80000` might work

mlfactory, 2019-02-07:
Hmm ok thx but here it look like it works https://github.com/letscontrolit/ESPEasy/issues/1050

I try it with a timer @10 sec.

Lets see if the wemos have after the boot a time :D

mlfactory, 2019-02-07:
Ok yes works perfect with a 30 sec timer.

I have seen if i reboot and login fast to easyesp the Local Time is for about 10-20sec 00:00:00 ;)

My Rule now:
```
On System#Boot do
timerSet 1,30 // 2 seconds timer

endon

on Rules#Timer=1 do
if %systime% > 08:00:00 and %systime% < 20:00:00
gpio,5,1 // relay off
endif
endon

```

That work.

Grovkillen, 2019-02-07:
Ok, Close this one

closed 2019-02-07 by Grovkillen