letscontrolit/ESPEasy #2291: Non volatile Custom parameters for usage in rules and http (feature request)

popaserge, 2019-02-07:
Actual custom variable are %v1% to %v16% >>>> Add custom parameters %p1% to %p16% working the same way but non volatile.

If easy to implement, this could be a way to wait for a more elaborated plugin (see issue #2284)

I discovered the Custom variables functionality and tried them in rules in combination with a dummy devices (using let command for the variables and TaskValueSet for the dummy devices)
Setting the Dummy device with a temperature set point and a device with a DS18b20 temperature sensor I was able to calculate 3 different temperature values:

Gap between temperature and set point
Sum of successive gaps
Variation between 2 successive gaps

and I was not so far to build a kind of PID (Proportional Integral Derivative) Time Proportioned output.

Just I needed to be able to set in the Dummy device the Temperature set point and 3 fixed parameters to be affected to the 3 calculated values and calculate a proportion of time to activate an output.

This is possible with http command but is not permanent as the Custom variable and the dummy device value are reset to 0 in case of reset or power down.

It could be interesting for similar application to have such a set of non volatile custom variables.

PPatou, 2019-02-07:
Yes indeed non volatile variables should be a must and open new possibilities. 
To be able to write a small pid with rules will give better results than just hysteresis.

TD-er, 2019-02-07:
I get it why a non volatile variable can be useful.
But how often will it be updated (and thus saved)?
We could ofcourse save it right before a reboot, but sadly not all reboots are intentional.

Saving it to flash may wear out the flash in no-time.

popaserge, 2019-02-08:
If used for example to build a small PID like I described, it just the time to find the good parameters.
I looked at this idea to see if it was possible to regulate the PH of a swimming pool more precisely than with the regulator device and its hysteresis parameter. 
I did the test on a temperature measurement because i have no way for the moment to measure a PH.
Of course to see if the good choice is done for the fixed parameters you need to monitor the result on the different parameters which are calculated periodically (Gap between temperature and set point,
Sum of successive gaps, Variation between 2 successive gaps)

By the way would you agree to give your opinion on issue #2227 which is an other part of what I checked for that idea.

closed 2026-08-17 by TD-er