letscontrolit/ESPEasy #2293: Networking:430:47: error: no matching function for call to 'UdpContext::listen(const ip_addr_t&, int)'

dony71, 2019-02-08:
compile mega release then result this 
compile with arduino ide 1.8.8
anything i'm missing?

> 

F:\arduino-1.8.8P250\ESPEasy\Networking.ino: In function 'bool SSDP_begin()':

Networking:430:47: error: no matching function for call to 'UdpContext::listen(const ip_addr_t&, int)'

   if (!_server->listen(*IP_ADDR_ANY, SSDP_PORT)) {

                                               ^

F:\arduino-1.8.8P250\ESPEasy\Networking.ino:430:47: note: candidate is:

In file included from sketch\ESPEasy-Globals.h:515:0,

                 from F:\arduino-1.8.8P250\ESPEasy\ESPEasy.ino:90:

F:\arduino-1.8.8P250\portable\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/UdpContext.h:98:10: note: bool UdpContext::listen(const ip_addr_t*, uint16_t)

     bool listen(CONST ip_addr_t* addr, uint16_t port)

          ^

F:\arduino-1.8.8P250\portable\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/UdpContext.h:98:10: note:   no known conversion for argument 1 from 'const ip_addr_t {aka const ip4_addr}' to 'const ip_addr_t* {aka const ip4_addr*}'

Networking:437:50: error: no matching function for call to 'UdpContext::connect(ip_addr_t&, int)'

   if (!_server->connect(multicast_addr, SSDP_PORT)) {

                                                  ^

F:\arduino-1.8.8P250\ESPEasy\Networking.ino:437:50: note: candidate is:

In file included from sketch\ESPEasy-Globals.h:515:0,

                 from F:\arduino-1.8.8P250\ESPEasy\ESPEasy.ino:90:

F:\arduino-1.8.8P250\portable\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/UdpContext.h:91:10: note: bool UdpContext::connect(const ip_addr_t*, uint16_t)

     bool connect(const ip_addr_t* addr, uint16_t port)

          ^

F:\arduino-1.8.8P250\portable\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/include/UdpContext.h:91:10: note:   no known conversion for argument 1 from 'ip_addr_t {aka ip4_addr}' to 'const ip_addr_t* {aka const ip4_addr*}'

Multiple libraries were found for "Servo.h"
 Used: F:\arduino-1.8.8P250\portable\packages\esp8266\hardware\esp8266\2.5.0\libraries\Servo
 Not used: F:\arduino-1.8.8P250\libraries\Servo
exit status 1
no matching function for call to 'UdpContext::listen(const ip_addr_t&, int)'

TD-er, 2019-02-08:
What version of the esp8266 arduino libraries are installed in your Arduino IDE?

amil109, 2019-02-13:
I have the same issue as @dony71. I'm using version 2.5.0 of the esp8266 arduino libraries.

I also tried compiling with older versions, but wasn't able to (other errors). Let me know if I can provide any other useful info?

TD-er, 2019-02-13:
You could add a define to the start of ESPEasy.ino
Something like this:
`#define CORE_2_5_0`

For PlatformIO I set these in the build parameters, since I only tested it with the dev/beta versions of core 2.5.0, there wasn't a define present to detect this version.
But now core 2.5.0 has been released I can add some proper code to the source to detect what core is being used.

So if you can test if that define is enough, then I can add proper changes to the code to make it compile in Arduino IDE without the need of user changes.

amil109, 2019-02-17:
Hi @TD-er - that seemed to have an effect. I no longer get the error above.

It's an interesting note that I was then able to compile without issue when I used 'Generic ESP8266 module' in board manager, but I got an error about iram space constraints when I used 'NodeMCU 1.0' (what I normally use).

Seems like the #define is enough to fix this specific error though - thank you (and sorry for the delayed response).

TD-er, 2019-02-17:
@amil109 Thanks for the feedback.
I will make some kind of define-work-around for the Arduino IDE builds.
And there are some issues with the iram space, but lots of them should already be resolved.

Do you have some other code using iram, which is not included in the normal dev tree?

amil109, 2019-02-17:
@TD-er - no, nothing extra - I was just looking to compile the latest branch with Arduino IDE so I could decode exceptions from my ESP8266... I'm wondering if I have a couple with bad flash, or if I've got a power supply issue (some chips reset every 5s or so, others are fine).

TD-er, 2019-02-17:
Maybe also have a look here (and the related issues) https://github.com/esp8266/Arduino/pull/5763

amil109, 2019-02-17:
Just to close on this one - it was a power supply issue!

I was able to build and flash the source from 16-Feb-19 without issue using the Arduino IDE after adding the #define above.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er