letscontrolit/ESPEasy #2295: Problems with MQTT import

jascdk, 2019-02-08:
Hi 

I have an issue with importing some temperature sensors into MQTT. I have a floor heating system that is connected to MQTT and reports the temperature via MQTT. This data will i import into ESP easy so i can get the areadings out on a oled display.

As for now, i can import up to 4 values - this works, but i have 11 sensors i would like to have data displayed from. If i add a new MQTT import then it just says 0 as values. 

How can i get this to work?

Grovkillen, 2019-02-08:
I suggest you use event values instead, no need to have them as proper values since they are only replicated from another source.

Grovkillen, 2019-02-08:
Concept of using eventvalues and internal variables:

```
On Update1 Do
 Let,1,%eventvalue1%
 Let,2,%eventvalue2%
 Let,3,%eventvalue3%
 Let,4,%eventvalue4%
EndOn

On Update2 Do
 Let,5,%eventvalue1%
 Let,6,%eventvalue2%
 Let,7,%eventvalue3%
 Let,8,%eventvalue4%
EndOn

On Update3 Do
 Let,9,%eventvalue1%
 Let,10,%eventvalue2%
 Let,11,%eventvalue3%
EndOn

On Rules#Timer=1 Do
 OLED,1,1,[VAR#1]
 OLED,1,6,[VAR#2]
 OLED,1,11,[VAR#3]
 OLED,2,1,[VAR#4]
 OLED,2,6,[VAR#5]
 OLED,2,11,[VAR#6]
 OLED,3,1,[VAR#7]
 OLED,3,6,[VAR#8]
 OLED,3,11,[VAR#9]
 OLED,4,1,[VAR#10]
 OLED,4,6,[VAR#11]
EndOn
```

And you publish those events from the source nodes.

closed 2019-02-08 by Grovkillen