letscontrolit/ESPEasy #2296: Weak Wifi signal on Release mega-20190202

sidhuneilkamal, 2019-02-08:
### Checklist
- [X ] The title of this issue is "to the point" and descriptive.
- [ X] This issue describes what is happening.
- [ X] This issue describes what components are affected (e.g. name of plugin/controller)
- [ X] This issue describes how to reproduce it.
- [ X] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2019-02-08:
There is NO description of what happens, not what setup you're using and no indication of what values you had before in what version.
Also no indication of what settings are used, what core library is used etc.
So please update the post or else I will close it.

sidhuneilkamal, 2019-02-08:
Thanks for reaching out, I am currently on 20190202. I was last on 20181008
and just updated yesterday to the most up to date. Is the 20190202 most
recent stable build? Also are their any settings in the dashboard that
would affect the WiFi signal? If so what are they and I will confirm.

TD-er, 2019-02-08:
What version of the firmware do you use?
In the ZIP file there are several builds.
So please give more information, like the info you get on the sysinfo page (use the Github copy button there)

sidhuneilkamal, 2019-02-09:
thank you for looking into this:
ESP Easy | Information |
 -----|-----|
Build:⋄|20103  - Mega|
Libraries:⋄|ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support|
GIT version:⋄|mega-20190202|
Plugins:⋄|46 [Normal]|
Build time:⋄|Feb  2 2019 03:13:22|
Binary filename:⋄|ESP_Easy_mega-20190202_normal_ESP8266_4096.bin|

Build:⋄
20103  - Mega
Libraries:⋄
ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
GIT version:⋄
mega-20190202
Plugins:⋄
46 [Normal]
Build time:⋄
Feb  2 2019 03:13:22
Binary filename:⋄
ESP_Easy_mega-20190202_normal_ESP8266_4096.bin

sidhuneilkamal, 2019-02-09:
before the update i was gretting -49db and now i am around -64 to -65db

jimmys01, 2019-02-09:
> 
> 
> before the update i was gretting -49db and now i am around -64 to -65db

This is still excelent signal. Perhaps core libs are giving diferent RSSI readings?

sidhuneilkamal, 2019-02-09:
In addition when I go to tools and select scan under WiFi. My wireless
shows at like 52db there. Yet on the home page it's at 65 to 66db

TD-er, 2019-02-09:
You could add a generic systeminfo plugin to sample the RSSI value and send it to some place where you can trace it and create a chart for it.
This way you can get an idea on the range of fluctuation of the signal and it will be always  with the same circumstances.

sidhuneilkamal, 2019-02-09:
So under logs I have uptime 486 and connection failures 48 wifi: 3
Under info I have 0 WiFi reconnects. Why am I getting so many mqtt
connection failures? Also like I said the WiFi dB was much better with the
older release.

TD-er, 2019-02-24:
In the upcoming nightly build there will be 2 distinct versions using core 2.6.0.
One is with SDK2.2.1 and one is using SDK3.0.0-dev.

The one with SDK3 will be very similar to what was "core 2.5.0" up-to a few weeks ago and SDK2.2.1 will be similar to what was used in core 2.4.2.
The switch back to the older SDK was mainly due to some issues with WiFi stability.

Could you test both builds?
N.B. these core 2.6.0 builds are based on what's now merged to the master branch of espressif8266/Arduino. So bugs may be fixed or added in that core library just about any new build we make.

sidhuneilkamal, 2019-02-25:
Which update is it? The 02/25?

TD-er, 2019-02-25:
Yep, the last nightly build has the core 2.6.0 in it with the 2 SDK versions.

sidhuneilkamal, 2019-02-26:
So i tried both bins and honestly they both felt the same to me - the rssi was still the same at 61 to 65. back in 181008 - i was in the 50's. So for now i just went with (ESP_Easy_mega-20190225_normal_ESP8266_4M)

TD-er, 2019-02-26:
They should be the same in reporting the RSSI.
If the more recent builds report a different RSSI, then we may have an issue here.
It is the strength observed by the wifi receiver of the signal.
Only thing is, it may fluctuate a lot, so to get a proper reading, we might need to take a number of samples and get some average (or modal) value of it.

Still I cannot explain why a build from 4 months ago was reporting a better RSSI given it is used under the same circumstances.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er