letscontrolit/ESPEasy #2299: 2 x DS1820b wont work proper

Sasch600xt, 2019-02-09:
Are there well knowing issues about using 2 DS1820B on 2 different IOs ?

In my case i am using one at IO 12 and one at IO 13.
So the one on IO 13 are not always sending data "NaN"
Sometimes it works, sometimes not.
I did changed the sensor and the port already. no luck.
When i have one configured and connected, all is fine.
But as soon the second comes in, i do not get always proper data from at least one of the sensors.

I did try with the 3 of the latest software version here.

pulltime for plugin is 60s and 59s. ( I thought it would help to seperate pullrequest by timing, but no luck so far)

Can someone help me out here and check it with own system ?
Bad thing is today i installed this for a heating system in a 1200qm hall.
So at the moment the heating system ist not working very well.

Thank you so much for your help !

Sascha

TD-er, 2019-02-09:
How are these Dallas sensors powered?

Sasch600xt, 2019-02-09:
powered by 3.3 volt and a 4,7k resistor to "+".
BUT ! I found it. Seems like gpio 12 and 13 are not very good for the sensors. The device reconnected nearly every hour. I used gpio 1 and 3 (RX/TX) and it works great. No problems anymore.
gpio 12 and 13 belongs to HW MOSI/MISO. So maybe thats not good.

TD-er, 2019-02-09:
Hmm, sounds more like you may need a pull-up resistor for those GPIO 12 and 13.  
For example  4k7 between the 3V3 and the GPIO pin.

Sasch600xt, 2019-02-09:
thats what i use always

jimmys01, 2019-02-10:
Why you need 2 different pins? The protocol they talk is called one wire for a reason

chunter1, 2019-02-10:
> Why you need 2 different pins? The protocol they talk is called one wire for a reason

Actually, "1-wire" is not primarily related to a bus system but the number of wires needed to communicate :)

RREE, 2019-02-10:
I can confirm the observed behaviour. I never managed to connect multiple 1w-sensors to more than one pin

Sasch600xt, 2019-02-10:
well....to be honest, now it is working great with gpio 1 and gpio 3. Just with gpio 12 and 13 i had big problems.

pwassink, 2019-02-13:
i disagree on problems with latest version and three ds12b20 sensors on a single gpio
have that working for a long time, even on the mega 20190202 version, core 2.4.1. 
and several with two of them, also no problems.

just to be sure, you've used only one pullup resistor on the combined "starnetwork" of ds18b20 wiring ?

uzi18, 2019-10-02:
@Sasch600xt @pwassink @chunter1 please paste here "debug" log from your dallas sensors readings on latest espeasy builds thanks

tonhuisman, 2023-09-03:
All 1-wire temperature sensor related issues seem to have been solved, so this issue can be closed.

closed 2023-09-03 by TD-er