letscontrolit/ESPEasy #2300: calls to NTP server

TungstenE2, 2019-02-10:
In my network I have Pihole installed. Today I recognized al lot of calls early in the morning.

I found ou that my ESPeasy used to controll my chicken house door did thousands of calls to the NTP server between 05:00 and 7:00. During the day only every hour.

What is the reason for this?

Installed version: mega-20190116 testing

 
[image]

[image]

TD-er, 2019-02-10:
Hmm, that's for sure not intentional.
It will only retry when it isn't able to get a proper reply.
My estimate is that about 1/3 of the reported NTP servers in the pool.ntp.org are offline, so it will retry when unsuccessful.
I thought I had put some limit on these retries, but apparently not.

ozett, 2019-02-13:
or are failover NTP-list/name (even if the first is a pool) after reaching the limit?

TD-er, 2019-02-13:
@ozett I don't understand your question, could you explain a bit more what you're thinking about?

TD-er, 2019-02-25:
Just to check, was this a one-time event, or did it happen again?

TungstenE2, 2019-02-25:
to be honest I did not check if it happend again

TungstenE2, 2019-02-25:
currently NTP is called every hour

[image]

TD-er, 2019-02-25:
That's just the DNS lookup right?
Still, it seems like a good indication for this issue.