letscontrolit/ESPEasy #2304: Signal filtering for noisy sensor signals

j-ghost, 2019-02-13:
### Checklist

- [ ] Temperature signals are very noisy, and jump up and down 4 degrees
- [ ] using a wemos d2 mini and a DHT11 temp-humidity sensor 

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable) But i did not find any info on this

### Summarize of the problem/feature request

Is it possible to create a feature so you can do some elementary basic signal processing on the sensor signals?

```

TD-er, 2019-02-13:
There is a rather elaborate filter option which you can already use now.
You can store values in a variable when using rules. Something similar can also be done using the dummy plugin.
These values can then be used in a computation (in the rules).
For example new value = (3* last_value + new sample) / 4
And then store that new value in the variable to be used in the next run as "last_value".

But I agree that some statistics module would be very useful.
On the other hand, if the values jump 4 degree between samples, then you can ask yourself how useful these are.
So maybe better to look into why these values are so unstable.

j-ghost, 2019-02-13:
Ah, what you describe is what i meant, but didn't realize it was there. maybe close the request than.
But i'm very new to this, so not sure how to do that precisely. I found the rule tab now. 
I copied the _new value = (3* last_value + new sample) / 4_ there but do not know how to put the _And then store that new value in the variable to be used in the next run as "last_value"._ to work.....

TD-er, 2019-02-13:
I will summon the rules guru by mentioning him ;)
@Grovkillen may be able to help here.
Or if it will end up in a discussion on how to do stuff instead of things that are buggy or need to change, then it may be best to move it to the forum.

But I guess he will be online again after 7 am tomorrow.

Grovkillen, 2019-02-14:
This is what I would try first. I store the last 9 values and add the newest value to that making it 10 values all in all. The average of those ten is then added to a dummy variable (task 12 in this example) which is then later used as the publishing task.

```
On Temp#Value Do
  Let,10,[VAR#9]
  Let,9,[VAR#8]
  Let,8,[VAR#7]
  Let,7,[VAR#6]
  Let,6,[VAR#5]
  Let,5,[VAR#4]
  Let,4,[VAR#3]
  Let,3,[VAR#2]
  Let,2,[VAR#1]
  Let,1,[Temp#Value]
  TaskValueSet,12,1,([VAR#1]+[VAR#2]+[VAR#3]+[VAR#4]+[VAR#5]+[VAR#6]+[VAR#7]+[VAR#8]+[VAR#9]+[VAR#10])/10
EndOn
```

Right at boot all the internal variables will be 0 making this rule not working great for the first ten loops.

closed 2019-02-14 by Grovkillen

TD-er, 2019-02-14:
Why use 10 samples? If you only use the a weighted average of the last reported value (not measured) combined with the last measured value, you will achieve similar filtering but a lot easier to implement in rules.
Only drawback is that you will still have some residue left of old values when the measured values suddenly change a lot. Taking the average of a fixed amount of samples will have no "memory" after N samples.

Grovkillen, 2019-02-14:
@TD-er Yeah I hear ya.

@j-ghost  You can try my filter and then also the filter suggested by TD-er:

```
On Temp#Value Do
  Let,2,[VAR#1]*3
  Let,1,[Temp#Value]
  TaskValueSet,12,1,([VAR#1]+[VAR#2])/4
EndOn
```

diegoflyer, 2022-12-16:
Guys no able to solve, the filter workw well:

On ds18b20#tempDS1 Do
Let,1,[ds18b20#tempDS1]
Let,2,[TempDDS1#Dummy]
TaskValueSet,5,1,(([VAR#1]+9*[VAR#2])/10)
EndOn

But I am trying to feed to the dummy device at boot the starting value unfiltered (if not done you will start from zero e you need time to arrive to the correct value). I add something like:

On TaskInit#ds18b20
Let,1,[ds18b20#tempDS1]
TaskValueSet,5,1,[VAR#1]
EndOn

Any idea? where am I wrong?

tonhuisman, 2022-12-17:
> 
> Any idea? where am I wrong?

You missed the required **do** in the first line of your added rule:
```
On TaskInit#ds18b20 Do
  Let,1,[ds18b20#tempDS1]
  TaskValueSet,5,1,[VAR#1]
EndOn
```

TD-er, 2022-12-17:
Since the last few builds (builds made after 2022/07/11), you have the option to use "stats". See the documentation on this: https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html?highlight=average#statistics
This allows you to take the average of the last N samples.
The advantage for this problem is that whenever you don't have enough samples, you effectively take the last sample.
This will allow you to "bootstrap" the filter.
Because you can set an arbitrary number of samples to use, you can easily tweak the filter.

diegoflyer, 2022-12-17:
@tonhuisman also with the missing do the value are not updated at boot! I have always 0 as starting value in the dummy
@TD-er average is an option but does not allow to filter in order to slow down sensor response on big time range
Any other suggestion?
i am not able to understand why the initial set of the value with TaskInit is not working....

TD-er, 2022-12-17:
> i am not able to understand why the initial set of the value with TaskInit is not working....

Because the value already has been set (but not read) when you process this event.
So on a reboot (power cycle), this value is still 0.

When using the average of the last N samples, you can still apply the filtering.

```
On ds18b20#tempDS1 Do
  Let,2,([ds18b20#tempDS1.avg2]*9+[ds18b20#tempDS1])/10
  TaskValueSet,5,1,[VAR#2]
EndOn
```

You play around with the nr of samples to take in the average.
Or even remove the last sample from the average before multiplying it with 0.9, then add 0.1x the new value.

diegoflyer, 2022-12-17:
@TD-er ok your suggestion give correct value at boot, bu anyway you can't work on low pass filter on big time range.
Maximum you can maage the buffer but ok could be enough for me!
I tried some oher solution to feed the starting value but no way:

`
on System#Boot do
  TaskDisable,5
  Let,5,0
  LogEntry,'disable task 5'
endon

On ds18b20#tempDS1 Do
  if [VAR#5] = 0
     Let,1,[ds18b20#tempDS1]-0.8
     TaskValueSet,5,1,[VAR#1]
     Let,5,1
     TaskEnable,5
  else
     Let,1,[ds18b20#tempDS1]-0.8
     Let,2,[TempDDS1#Dummy]
     TaskValueSet,5,1,(([VAR#1]+9*[VAR#2])/10)
  endif
EndOn
`

TD-er, 2022-12-17:
Not sure if task 5 on your setup is a dummy or not.
If it isn't a dummy, you may want to make sure to allow it on the Tools->Advanced page to use TaskValueSet on others than a dummy task.

Also I'm not 100% sure calculations will work when used in taskvalueset as an argument.
If not, then you either may want to add an `=` sign before the calculation, or perform all calculations in the `Let` command.

Also a dummy task should be "run" using taskrun, or else it will not send out the task values to the connected controllers.

Not sure why you can't work on low pass filter on a big time range with my suggested code.
This should be no different than you had before:

```
On ds18b20#tempDS1 Do
  Let,2,([var#1]*9+[ds18b20#tempDS1])/10
  TaskValueSet,5,1,[VAR#2]
  Let,1,[ds18b20#tempDS1.avg2] // store avg for next run
EndOn
```

diegoflyer, 2022-12-18:
@TD-er 
my code was different because I was working with the value in the dummy device not calculating on the same "source" task.
I did not know that was possible to force the value in the same "source" task, I will have a try.
Now I have this solution that looks working:

`on System#Boot do
  Let,3,0
endon

On ds18b20#tempDS1 Do
  if [VAR#3] = 0
    Let,1,[ds18b20#tempDS1]-1
    TaskValueSet,5,1,[VAR#1]
    Let,3,1
  else
    Let,1,((([ds18b20#tempDS1]-1)+9*[TempDDS1#Dummy])/10)
    TaskValueSet,5,1,[VAR#1]
  endif
EndOn
`

thalesmaoa, 2025-06-20:
Just a small contribution

This is a simple but effective 2nd-order Butterworth low-pass filter, more accurate than a moving average.

It runs entirely in ESPEasy Rules, using internal variables (Let,<n>,<value>).

Two cutoff frequencies are available:

    fc = 0.2 Hz when v8 = 0
    fc = 0.1 Hz when v8 = 1

All parameters were calculated assuming a 1-second sampling interval (i.e., one event per second).
If you can trigger events at a faster rate, new filter coefficients must be calculated.

```
on input1#ai1 do
  Let,8,1       // Set filter mode: 1 = 0.1 Hz, 0 = 0.2 Hz
  Event,filt
endon

on filt do
  // Shift input history
  Let,4,[VAR#3]       // x2 ← x1
  Let,3,[VAR#2]       // x1 ← x0
  Let,2,[input1#ai1]  // x0 ← new input

  // Shift output history
  Let,7,[VAR#6]       // y2 ← y1
  Let,6,[VAR#5]       // y1 ← y0

  // Compute filtered value
  If %v8%=1
    Let,5,0.067455*%v2%+0.134911*%v3%+0.067455*%v4%+1.14298*%v6%-0.412802*%v7%
  Else
    Let,5,0.2066*%v2%+0.4132*%v3%+0.2066*%v4%+0.3695*%v6%-0.1958*%v7%
  EndIf
endon

```

TD-er, 2025-06-20:
@tonhuisman Can you incorporate this rules example in one of your "docs update" PRs?