letscontrolit/ESPEasy #2307: Espeasy Whether to default internal pull-up EN(CH PD)

blackhu, 2019-02-14:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2019-02-14:
Not sure if you're still editing the start post?
At least I don't get the issue right now.

TD-er, 2019-02-14:
Just to clarify why I don't understand this issue.
The EN (CH_PD) pin is a pin to effectively switch on the ESP.
So on most boards it is either directly tied to Vcc or via some resistor.
It may be possible to use it as some delay to make sure to only boot the ESP when the power is stable enough (e.g. when a capacitor needs to be charged via some solar panel or something like that)

But as far as I know, there is nothing we can do from software to enable this pin, since it is actually booting the ESP.

Please correct me if I'm wrong.

blackhu, 2019-02-15:
I just do'nt like use connection to vcc do pullup. so can i try ask the question

blackhu, 2019-02-15:
> Just to clarify why I don't understand this issue.
> The EN (CH_PD) pin is a pin to effectively switch on the ESP.
> So on most boards it is either directly tied to Vcc or via some resistor.
> It may be [...]

closed 2019-02-15 by blackhu

TD-er, 2019-02-15:
But how should we make it different in software?
The EN pin does start the ESP, so before that we simply cannot do anything in software.
Also the pull-up or connection to Vcc is made by the one designing the board you're using.

So as far as I know, there is nothing we can do in software to change it.

blackhu, 2019-02-18:
think you