letscontrolit/ESPEasy #2309: Mozilla WebThings API

blackfyre, 2019-02-15:
Hi!

Love the effort you've put into making the our favorite ESPs even more accessible!

I was wondering if you would entertain the thought of implementing the Mozilla WebThing API, there's already an open ticket (#703) to make tings more API like, and this could provide an option for that ;)

The API description can be found at: https://iot.mozilla.org/wot/

The guys at Mozilla are really putting an effort into creating this communication standard and their gateway has a solid design.

TD-er, 2019-02-15:
I haven't read their API yet.
Is this a request for a controller (in ESPeasy terms) or change of webserver?
Or should this API be used to fetch data from elsewhere (thus more like a plugin)?

blackfyre, 2019-02-15:
Thanks for taking your time on this!
The API is meant to both read statuses/sensors and to control devices. I've done few implementations on both cases for the ESP8266:
 * https://github.com/blackfyre/sonoff-basic-webthing
 * https://github.com/blackfyre/ds18b20-webthing
The API is aimed at communicating with Mozilla's Gateway application, but since it's an open standard (draft, atm), other systems like Home Assistant/Domoticz/OpenHab/... are free and even encouraged to implement it on their and as recommendations from Amazon and/or Google leave some things to be desired, while other manufacturers are set on their closed solutions.