letscontrolit/ESPEasy #2310: Newest Release 20190215 without wroom Core 250?

fraeggle, 2019-02-15:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

fraeggle, 2019-02-15:
Newest Release 20190215 without Wroom-02 Core 250? What happend to the 250 Core, because with the 250 Core it uses the 2Mb not only 1Mb like in 242, so with this one no OTA Update is possible.

Grovkillen, 2019-02-15:
We're experiencing problems with the 250 core and have removed it from the nightly until it's stable.

closed 2019-02-15 by Grovkillen

TD-er, 2019-02-15:
That's correct, the last few builds I made myself based on Core 2.5.0 were experiencing crashes when accessing some web pages, so I disabled the core 2.5.0 builds until those are resolved.

fraeggle, 2019-02-15:
Thank you for the info.. So i will wait...
@TD-er : Hmm any special web pages? Because haven't noticed any Problem..
Regards Peter

TD-er, 2019-02-15:
Well, not specifically a single page, but on 2 nodes it was reproducible to open the controller settings or the advanced settings page.
It isn't a problem on all nodes, so it seems to be somewhat hardware specific.

fraeggle, 2019-03-01:
Don't want to disturb... I read in Release 20190225 [Build] Use PlatformIO Espressif8266@2.0.1 (core 2.5.0).
i thought in CORE 2.5.0 the whole memory is used then. When I load the current build (ESP_Easy_mega-20190301_normal_WROOM02_2M.bin) into my Wroom2, it formats only 1MB again.
With the beta (core_250_beta_WROOM02_2M256) all the memory was used. Is this a setting thing?

TD-er, 2019-03-01:
The 2 MB layout (1.75 MB for sketch + OTA & 256k for SPIFFS) we used was something that was added in core 2.5.0.
So maybe we should mark that WROOM02 build to have different partition layout to make it more clear.
This layout is not usable for older core builds.

fraeggle, 2019-03-01:
hmm  haven't seen this label (`normal_core_250_WROOM02_2M256` (1.75M sketch & OTA, 256k SPIFFS))
in the bin folder..

TD-er, 2019-03-01:
That's right. I have taken them out of the nightly builds, since they were causing all kinds of issues.
That version is present in the platformio.ini file though, so you can build a version yourself.

2 days ago I changed some settings and also PlatformIO released some fixes and set the lib version to `
Espressif8266@2.0.1` (.1 version) so we can try if it is now working again.
Then I can enable those core 2.5.0 builds again to be included in the ZIP file.

fraeggle, 2019-03-01:
thanks a lot. i can test if you want to. i have 15 of them in use ... ;-)  and very satisfied